Bodmer/TFT_eSPI #314: ili9486 for uno and esp32

joe318, 2019-02-20:
Hello,

I have the ili9486 for uno in shield mode, i'm trying to use ur library with the screen but nothing appear :(
I try thousands things without success, any suggestion?
it's one waveshare for arduino not rpi.

Regards,

Bodmer, 2019-02-20:
Are you using an UNO format ESP32.  If so have you make the link modifications?

joe318, 2019-02-20:
No, im using one normal esp32.

[image]

joe318, 2019-02-20:
But i'm not the only one to have this problem with specially this tft.

[image]

Bodmer, 2019-02-20:
The display is a 16 bit SPI interfacebased around shift registers so it may not be compatible with TFT_eSPI.

How have you wired the display to the ESP32?

Post your setup file so I can see all the settings you are using.

joe318, 2019-02-21:
ok, look, i search in your issues and i found a guy with the same tft and same problem, i read ur answers, so i decide to stop spend time but i spend money and i buy one for rpi version loool, so dont worry, i leave this one for play with an arduino.
But thanks coz u answer fast and u are fucking smart lol

closed 2019-02-21 by Bodmer