Bodmer/TFT_eSPI #315: Distorted text

TBKRK, 2019-02-21:


TBKRK, 2019-02-21:
Hi, using printToSprite(X), while textWidth(X) is greater than 256, produces distorted printout. I there any limit while printing to Sprite?

Bodmer, 2019-02-21:
There is no intentional limit to 256. The Sprite will need memory and so it may be the case that there is not enough RAM left to create a Sprite.

Are you using an ESP8266 or ESP32?

Can you provide a minimal sketch that demostrates the problem that I can test?

TBKRK, 2019-02-21:
Hi, thank you for response.

It is ESP32

Here is the code, with added line to print text and show the problem and screenshot (see the 4th and 5th word).

 

void GraphicDisplay( int Index)

{

#define XMAX 320

#define LINESPACE 8

#define LEFTMARGIN 15

#define RIGHTMARGIN (XMAX - LEFTMARGIN)

#define RIGHTBISMARGIN 218

int YPos = 3, XPos = 0, t, k, l;

String Temp1, Temp2;

 if (Results[Index].PM25 == 0 && Results[Index].PM10 == 0) return;      

 

tft.setTextColor(TextCol, BackgroundCol); 

                                         

 tft.setCursor(0, 0); 

 

tft.loadFont(AA_FONT_LARGE); 

 spr.loadFont(AA_FONT_LARGE); 

 

spr.setTextColor(HeadCol, BackgroundCol);

 Temp1 = " " + Results[Index].Name + " ";

 

tft.setCursor(XMAX / 2 - tft.textWidth(Temp1) / 2, YPos);    //XMAX / 2 - spr.textWidth(Temp1) / 2

spr.printToSprite(Temp1);

YPos += tft.fontHeight() + LINESPACE;

tft.fillRect (0, 3, XMAX / 2 - tft.textWidth(Temp1) / 2 , YPos - 3, BackgroundCol);

tft.fillRect (XMAX / 2 + tft.textWidth(Temp1) / 2    , 3, 320 , YPos - 3, BackgroundCol); 

 

 tft.loadFont(AA_FONT_SMALL); 

 spr.loadFont(AA_FONT_SMALL); 

 

Temp1 = " " + Results[Index].Address + " ";

if (Results[Index].Address != "Lokalizacja nieznana" && Results[Index].Dist > 0.01 && Temp1.length() < 40 ) Temp1 = Temp1 + "(" + Results[Index].Dist + " km) ";

 

Temp1 = "Kraków Kraków Kraków Kraków Kraków ";  // JUST TO DEMONSTRATE PROBLEM !!!

 tft.setCursor(XMAX / 2 - tft.textWidth(Temp1) / 2, YPos);   

 spr.printToSprite(Temp1);

 tft.fillRect (0, YPos , XMAX / 2 - tft.textWidth(Temp1) / 2 ,  tft.fontHeight() , BackgroundCol); 

 tft.fillRect (XMAX / 2 + tft.textWidth(Temp1) / 2    , YPos, 320 , tft.fontHeight(),  BackgroundCol); 

 

YPos += tft.fontHeight() + LINESPACE;

 

tft.loadFont(AA_FONT_LARGE); // Must load the font first

spr.loadFont(AA_FONT_LARGE); 

 

spr.setTextColor(Pm25Color(Results[Index].PM25), BackgroundCol);

 Temp1 = " " + Float2String(Results[Index].PM25, 1) + " ";

  tft.fillRect (84, YPos , RIGHTBISMARGIN - tft.textWidth(Temp1) -84 ,  tft.fontHeight() , BackgroundCol); 

 tft.setCursor(RIGHTBISMARGIN - tft.textWidth(Temp1), YPos );   

 spr.printToSprite(Temp1);

YPos += tft.fontHeight() + LINESPACE;

 

spr.setTextColor(Pm10Color(Results[Index].PM10), BackgroundCol);

Temp1 = " " + Float2String(Results[Index].PM10, 1) + " ";

tft.fillRect (84, YPos , RIGHTBISMARGIN - tft.textWidth(Temp1) -84 ,  tft.fontHeight() , BackgroundCol);  

 tft.setCursor(RIGHTBISMARGIN - tft.textWidth(Temp1), YPos );  

 spr.printToSprite(Temp1);

YPos += tft.fontHeight() + LINESPACE;

 

spr.setTextColor(TextCol, BackgroundCol);

Temp1 = " " + Float2String(Results[Index].Temp, 1) + " ";  

 tft.fillRect (104, YPos , RIGHTBISMARGIN - tft.textWidth(Temp1) -104 ,  tft.fontHeight() , BackgroundCol); 

 tft.setCursor(RIGHTBISMARGIN - tft.textWidth(Temp1), YPos );   

 spr.printToSprite(Temp1);

YPos += tft.fontHeight() + LINESPACE;

 

Temp1 = " " + Float2String(Results[Index].Hum, 1) + " ";

tft.fillRect (104, YPos , RIGHTBISMARGIN - tft.textWidth(Temp1) -104 ,  tft.fontHeight() , BackgroundCol); 

 tft.setCursor(RIGHTBISMARGIN - tft.textWidth(Temp1), YPos );   

 spr.printToSprite(Temp1);

 

 tft.unloadFont(); 

 spr.unloadFont();

 

}

 

 

Regards - Tomasz

Bodmer, 2019-02-22:
I need a complete self contained example, ideally in a zip file, potentially this could be based on an existing library example.

The code you posted will generate compile errors that I do not want to sort out!

TBKRK, 2019-02-22:
Hi, here is complete scratch, with library setup file, fonts and picture.

 

Regards - Tomasz

Bodmer, 2019-02-22:
Hi,
Thanks, I understand the problem. It is the [same one repoorted here](https://github.com/Bodmer/TFT_eSPI/issues/311).

This has been fixed in the master library here on Github. Try loading this updated version 1.4.8 and post back if you still have problems.

Thanks for reporting this.

TBKRK, 2019-02-22:
Thanks I’ll check…

TBKRK, 2019-02-22:
Hi, 1.4.8 solved the problem, thank you!

 

Regards – Tomasz

Bodmer, 2019-02-22:
Great.

closed 2019-02-22 by Bodmer