Bodmer/TFT_eSPI #316: DrawRect in sprite causes ILI9486 reset

armageddon421, 2019-02-22:
Hey,

When trying to drawRect in a sprite in some scenarios the display is resetting (turns white, won't draw anything anymore). I could fix it by overriding the drawRect funtion also in the sprite class, removing the begin and end SPI calls. They probably somehow caused the reset and aren't needed anyways when drawing into a sprite.
Would you like a pull request for this?

Greetings!

Bodmer, 2019-02-25:
There was a bug in the range checking a while ago that may have caused the problem. This was fixed, so please try the latest Github master.

armageddon421, 2019-02-28:
I'm on the latest master already.
The issue does not always happen, it started happening at some point when I extended my code. Getting rid of the SPI calls fixed the issue, that's why I suggested adding it, just to be sure.

armageddon421, 2019-02-28:
Maybe override spi_begin and end in the Sprite class so they don't do anything? That would fix it for all cases where drawing funtions from the base class are reused.

Bodmer, 2019-03-04:
I suspect you are running out of memory or running off the end of an array as extending the sketch does not change the behaviour of the TFT_eSPI library drawRect() function.

Making spi_begin and end virtual and over-riding in the Sprite library will mean those functions can no longer be inline and thus there will be a performance penalty for the SPI rendering. I think there is an easier way by making use of the existing semaphore flags. I will think on this further and update the library at the next release if I find a neat solution.

closed 2019-03-04 by Bodmer