Bodmer/TFT_eSPI #317: ili9486 and nodemcu

joe318, 2019-02-23:
Hello mr Bodmer,

I buy the 3.5 inch rpi version, i think it's the ili9486, not sure at 100% but 99%.
I try manyyyyy different configurations like reduce the spi, i try allll the setup ili9486, i disconnect the touch, and disconnect and connect everything maybe 10 times, but i always have the same result, i put you pics for you see.
[image]
[image]
[image]

Can i have help to find where is my problem?
Regards,

joe318, 2019-02-23:
Just another test, when i choose Setup6_RPi_Wr_ILI9486.h, the fonts work better, not good too but better.

joe318, 2019-02-23:
Another important detail, its a waveshare rev A ver 3 for rpi

joe318, 2019-02-23:
after another tests, work well with raspberry and esp32 but with schreibfaul1 library, i want put work with the 8266 :(

Bodmer, 2019-02-25:
Try the latest version of the library, there was a bug associated with the RPi display that shloud be fixed in the latest version.

joe318, 2019-02-25:
oh cool, i'm waitingggggg lol, when you think the library can be ready?
kiss

Bodmer, 2019-02-25:
The library was updated 1 week ago in response to issue #310 which looks like the same problem. Download the latest from Github and try it.

closed 2019-03-04 by Bodmer

joe318, 2019-03-20:
Hello, was try with one wemos and nodemcu, esp8266 but always the same problem, i try many things like what they do in the issue 310 but without success :(

joe318, 2019-03-20:
The most strange it's the better solution i found (still not perfect) it's when i use yout setup6, can see the text nice but not the background with the rainbow example for example.

joe318, 2019-03-21:
ok i found why, he work well with the SPI_FREQUENCY  10000000 shit it's slow no?

Bodmer, 2019-03-21:
Thanks for reporting the solution. You may find it works OK at 16MHz, try:

SPI_FREQUENCY 16000000

joe318, 2019-03-21:
hello, tested now at 16000000, better than 20 but not perfect

joe318, 2019-03-21:
i test now at 15, look working, need try more

joe318, 2019-05-18:
Mr bodmer, just for help people, the 4inch waveshare for uno work with my nodemcu with the spi speed at 15
if can help pp