Bodmer/TFT_eSPI #318: Problem with Create_font.pde script

TBKRK, 2019-02-23:
Hi, there is a small problem displaying Polish characters, using printToSprite(). Characters like "Ż Ó Ź Ć" are too high and the upper part is cut out.
[test.zip](https://github.com/Bodmer/TFT_eSPI/files/2896998/test.zip)

Bodmer, 2019-02-25:
Unfortunately the Processing IDE uses the top of the "d" for the maximum glyph height, so this is why the top is lost.

I commented out code here which looked for the highest character as that gave stupid results if non-existent characters are included in the font...

You can adjust the font metrics immediately after loading the font thus:

```
void GraphicDisplay()
{
#define XMAX 320
#define AA_FONT_SMALL "times36"

  String Temp;

  spr.setTextColor(TFT_YELLOW, TFT_BLACK);

  spr.loadFont(AA_FONT_SMALL);

  if (spr.fontLoaded)
  {
    spr.gFont.maxAscent = spr.gFont.maxAscent + 8;
    spr.gFont.yAdvance = spr.gFont.maxAscent + spr.gFont.maxDescent;
  }

  Temp = "ż Ż ó Ó ź Ź ć Ć";

  tft.setCursor(XMAX / 2 - tft.textWidth(Temp) / 2, 100);
  spr.printToSprite(Temp);

  spr.unloadFont(); // Remove the font to recover memory used

}
```
Not that the font does not need to be loaded in the the tft class as well, so this function has been edited.

In this example an extra 8 pixels has been added to the height.

I will think further on this issue to see if there is a better way to deduce or set the font height.

TBKRK, 2019-02-25:
Hi, thank you for email. Mentioned capital letters are not used too often, it is not a serious problem.

 

Regards, Tomasz

closed 2019-02-25 by Bodmer