Bodmer/TFT_eSPI #320: RTL Support

gilshahar7, 2019-02-27:
Hey, im trying to display Hebrew text using tft.print.

I was able to create a custom font to actually display the Hebrew letters but displaying them will result in a reversed string. 
For example: when i try to write "עברית" it will show up as "תירבע" because it takes the characters in a regular order and prints each one from left to right.

Now i was able to reverse the strings that i want to display, but another issue came by.
if the string needs to be more than one line, it will start at the lower line instead of the first line. (makes sense).
Example: the string is `"בדיקה של משפט ארוך בשביל לראות אם המסך עובד"`
and it will display like that:

```
ביל לראות אם המסך עובד
בדיקה של משפט ארוך בש
```
I was wondering if there could be a way to add an option to declare writing from right to left before printing the text. that way i wont need to flip the text and it will be in the correct order when printed.

Thank you very much.

Bodmer, 2019-02-27:
Until RTL handling is incorporated then the simplest approach is to position the text using coordinates by setting the text datum to top-right and then using drawString():

```
  tft.setTextDatum(TR_DATUM); // Top Right datum
  tft.drawString("שלום", tft.width()-1, 0);
```

You also make sure that the string is not too long by checking the string width in pixels and splitting the string up in the sketch if it is too long.

```
String message = "שלום";
uint16_t pixel_width = tft.textWidth(message);
```

gilshahar7, 2019-02-28:
I tried what you suggested but its the same thing just aligned to the right. the letters are still flipped.

I'll wait for proper RTL support because i need to write english text inside hebrew text which is more complicated to do right now.

Bodmer, 2019-02-28:
To check my understanding, you wish a string:  "ABCD"

To  be printed to screen right justified as:    "DCBA"

With the cursor moving right to left as more characters are printed?

gilshahar7, 2019-02-28:
Yes, in Hebrew the first letter of a sentence/word is in the most right
position and every character is displayed to the left.

When I try to write "שלום" the string structure is:
[0] 'ש'
[1] 'ל'
[2] 'ו'
[3] 'ם'

 And on the screen it prints out "ש" as the first letter which is fine but
the second letter is "ל" and it gets written to the right of the first
letter instead of the left so we end up with "םולש"

There needs to be a way to tell the cursor to move to the left after every
character it prints instead of to the right.

gilshahar7, 2019-02-28:
http://unicode.org/reports/tr9/

This is an article about Bidirectional typing. maybe it can help you understand.

gilshahar7, 2019-02-28:
If the cursor direction can be modified i can try to even incorporate english words inside hebrew sentences.

I'll just need to divide the sentence into sections depending on what direction they need to be.
And then i can define how to write them, from RTL or LTR, and of course i will need to calculate the width of the sections and move the cursor over so that the words will not overlap.

I think its called Bidirectional typing.

Bodmer, 2019-02-28:
The Unicode standard defines an approach for handling bi-directional text and controlling the writing direction:

http://www.unicode.org/reports/tr9/

This would be complicated to implement fully and I am not sure how often these features will be needed. So I have come to the conclusion that any RTL future addition would most likely be in a companion library. so the the TFT_eSPI library just handles the low level graphics rendering. There may already be code available on Github so if anyone wished to pick up the challenge of implementing the Unicode approach then please post here.

Bodmer, 2019-02-28:
Another potentially useful link for information:
https://www.w3.org/TR/css-writing-modes-4/#propdef-text-orientation

closed 2019-02-28 by Bodmer

Bodmer, 2019-02-28:
I see we crossed posts, I will think on this further as I suspect a simple RTL text rendering method could be handled in the sketch within a function by maintaining the cursor position in the sketch code and using the existing rendering support in TFT_eSPI.

gilshahar7, 2019-02-28:
Yes, if a simple RTL cursor support will be added i might be able to complete the rest. i already made myself a whole function to print Hebrew by flipping the text, then dividing into lines and printing them upside down which works fine on Hebrew only text.
but if ill be able to control whether the text is printed from RTL or LTR ill be able to mix English and Hebrew in the same string and it will make the function less power hungry.

Bodmer, 2019-03-01:
Try this:

[drawStringRTL.zip](https://github.com/Bodmer/TFT_eSPI/files/2917517/drawStringRTL.zip)

gilshahar7, 2019-03-01:
Nice! seems to work great with hebrew, ill try to make something out of it to mix RTL & LTR with numbers and make it look normal.

Bodmer, 2019-03-02:
I have added an drawStringLTR() to the sketch so that the line wrapping and cursor can be used to mix RTL and LTR characters in the same line:

[drawStringRTL_drawStringLTR.zip](https://github.com/Bodmer/TFT_eSPI/files/2921607/drawStringRTL_drawStringLTR.zip)

Bodmer, 2019-03-04:
I have added support for RTL rendeing in the TFT_eFEX library, [see example here](https://github.com/Bodmer/TFT_eFEX/tree/master/examples/drawStringRTL).