Bodmer/TFT_eSPI #322: ST7789 240*240 RED and BLUE swap and only work when define #define ILI9341_DRIVER

Kaitoukid93, 2019-03-03:
Hello, thanks for the library
Im facing a weird problem with st7789 240*240 ips
if I define chip model as st7789 in user-setup.h the screen remain black with color test sketch, regardless the settings (reset pin =-1, height and width defined 240, both test with and without rotation)
if I define chip model as ILI9341 in user-setup.h the screen display ok with these settings:
-rotation(2)
-comment out reset=-1( even I connect LCD reset pin with node mcu reset)
-color invert set to ON
please help me

Kaitoukid93, 2019-03-03:
and one more issue I I found with smooth font that I cant use any custom *vlv font that I created with the processing tool, the sketch data upload successfully and only default font (that you already included in the data folder) display, my font (that created with processing and coppied to the same datafolder) wouldnt show and the serial monitor show:

SPIFFS available!

Fonts found OK.
Font file Agency FB72.vlw not found!

Bodmer, 2019-03-04:
I suspect SPIFFS does not allow spaces in file names. you can't use underscore either, use - instead.

Did you [define the width and height](https://github.com/Bodmer/TFT_eFEX/tree/master/examples/drawStringRTL)?

Kaitoukid93, 2019-03-04:
Yes! I did, both case ( ILI9341 and ST7789) I defined 240*240 in user setup
and I also try change the name of the file but same result
SPIFFS did find the font because Fonts found OK
"font file not found!" is not the warning that declared in the sketch, it's in the library I think

Bodmer, 2019-03-04:
OK, I will investigate.  The ST7789 files were updated via a pull request from a library user a while ago so there may be an issue there.

Bodmer, 2019-03-04:
Yes, there is an issue introduced by the pull request a while ago:

https://github.com/Bodmer/TFT_eSPI/pull/182/commits/8eefc6f26e77710973a3a414bf963749b97cf346

I will make a change and hopefully that will fix one of the issues regarding the colours.

I will investigate the font file issue. Can you provide a minimal example as a zip file that demonstrates the problem?

Kaitoukid93, 2019-03-04:
I will give you the original OTF font and the Converted vlw, the sketch is demo font example
https://drive.google.com/open?id=1_X3MipcSQ5UiriMq1KUU550PPx_bFJbM

Bodmer, 2019-03-04:
Are you using ESP32 or ESP8266?

Kaitoukid93, 2019-03-04:
I'm using ESP8266. thanks

Bodmer, 2019-03-04:
I adapted the Font_Demo_2 example to use the "BIG JOHN72" you provided and it is working fine on both ESP32 and ESP8266.  Try this:

[Font_Demo_2_issue_322.zip](https://github.com/Bodmer/TFT_eSPI/files/2928602/Font_Demo_2_issue_322.zip)

Kaitoukid93, 2019-03-04:
Ah yes my bad,
I was define the font with its extension as below
#define AA_FONT_LARGE "BIG JOHN72.vlw"
sorry
I still cant fix the st7789 model define, still black screen

Bodmer, 2019-03-05:
This is the setup I used on a 240x240 ST7789 display with an ESP32. You should only need to change it to the pin numbers you use:

```
#define ST7789_DRIVER

#define TFT_SDA_READ

//  #define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
//  #define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red

#define TFT_WIDTH  240 // ST7789 240 x 240 and 240 x 320

#define TFT_HEIGHT 240 // ST7789 240 x 240

#define TFT_INVERSION_ON
// #define TFT_INVERSION_OFF

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS    15  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

#define SPI_FREQUENCY  40000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
// #define SPI_TOUCH_FREQUENCY  2500000

// #define SUPPORT_TRANSACTIONS
```
These are the pins I used for the ESP8266:
```
// Display SDO/MISO  to NodeMCU pin D6 (or leave disconnected if not reading TFT)
// Display LED       to NodeMCU pin VIN (or 5V, see below)
// Display SCK       to NodeMCU pin D5
// Display SDI/MOSI  to NodeMCU pin D7
// Display DC (RS/AO)to NodeMCU pin D3
// Display RESET     to NodeMCU pin D4 (or RST)
// Display CS        to NodeMCU pin D8 (or GND)
// Display GND       to NodeMCU pin GND (0V)
// Display VCC       to NodeMCU 5V or 3.3V

#define TFT_CS   PIN_D8  // Chip select control pin D8
#define TFT_DC   PIN_D3  // Data Command control pin
#define TFT_RST  PIN_D4  // Reset pin (could connect to NodeMCU RST, see next line)
```
Have you tried any other examples?

Kaitoukid93, 2019-03-05:
I've use the same setup for esp8266 as your post, still black screen after upload
TFT_CS is missing so I commented it
TFT_DC to pin D3 (GPIO 0)
TFT SCK to pin D5 (GPIO14)
TFT_SDA to pin D7 (GPIO 13)
reset pin to reset pin of esp8266
with the same setup, I uncommented first line (IL9341_DRIVER), the screen show ok but RED and BLUE swap, rotation(2)

Bodmer, 2019-03-05:
This indicates the init code is not working wth your display. There are diferent versions of the display chip but I would expect them to use the same init  code.

Can you post a picture of the front andback of your display and a link to the sellers web page.

Bodmer, 2019-03-05:
I have tried using the ILI9341 driver with my ST7789 249z240 display and it does not work, the display remains completely blank.  This demonstrates your display is different to mine in some way.

It may be the driver chip version or it may be a completely different driver chip in yours.  So post photos of your display.

Kaitoukid93, 2019-03-05:
here. I have the sample for you. it looks like the DSTIKE ST7789 1.3 inch that I saw a pull request before
https://farm8.staticflickr.com/7907/32346186177_b33c42c0a6_o.jpg

Bodmer, 2019-03-06:
Your display looks identical to mine. The only difference remaining is that I am using an ESP32 for my tests. I will try connecting my display to an ESP8266 and see if the results are different.

Bodmer, 2019-03-07:
I have come to the conclusion that the display is not being reset correctly when the ESP8266 reset pin is used as for a while the display receives garbage until the SPI bus is stable.

In the next few days I will add further delays at start up and always add a software reset.

orhanyor, 2019-03-22:
hey guys, anyone tried this library with M0 based boards? i got one coming my way i wonder if i can use this library for ST7789 240*240 TFT. thanks

Bodmer, 2019-03-23:
This library is for ESP8266 and ESP32 only. It uses low level register calls for performance reasons, and these calls are not compatible with the M0.

Kaitoukid93, 2019-03-25:
I'm still waiting for this fix. 
I can use IL9341_DRIVER at the moment but cant fix the red and blue swap and the rotation must be 2
thanks

achillhasler, 2019-03-29:
for the red / blue switch, set in ILI9341_Rotation.h at the top
#undef TFT_MAD_BGR
#define TFT_MAD_BGR 0

change ILI9481_Init.h
    writecommand(TFT_MADCTL);
--    writedata(0x0A);
++  writedata(0x00);
and wait til bodmer has fixed the st7789 code

Bodmer, 2019-04-08:
I have tried various setups and have found that these displays need to have the reset pin (RES) driven by a hardware pin otherwise initialization is not reliable. The problem is that with CS tied low there is no start of clock signal and thus spurious clocks at start up upsets the display.

I there fore suggest that you try a hardware reset signal. If that does now work then unfortunately I have no solution.

Shockwave2k, 2019-05-30:
seems to work with your latest suggestion.
[image]

used a lolin nodemcu v3 & this as a display from [aliexpress.com](https://de.aliexpress.com/item/1-3-zoll-IPS-HD-TFT-ST7789-Stick-IC-240-240-SPI-Kommunikation-3-3-V/32880846744.html?spm=a2g0s.9042311.0.0.3e684c4dodjYEp)

[configs.zip](https://github.com/Bodmer/TFT_eSPI/files/3239098/configs.zip)

closed 2019-07-05 by Bodmer