Bodmer/TFT_eSPI #325: Dont get ILI9341 to work on esp8266

marcvs, 2019-03-07:
Hi There,

I'm using the ILI9341 with an ESP8266.
Pins and general functionality are proven to work with the Adafruit_ILI9341 graphicstest example.

I tried a few TFT_eSPI examples but all result in a flickering display.

In User_Setup.h I defined these values:
```
#define TFT_CS   16 // PIN_D0
#define TFT_DC   15 // PIN_D8
#define TFT_RST  5  // PIN_D1

#define TFT_MOSI 13
#define TFT_CLK  14
#define TFT_MISO 12
```
I also played with frequencies -- no success.

The example Read_User_Setup returns this:
[code]
TFT_eSPI ver = 1.4.5
Processor    = ESP8266
Frequency    = 80 MHz
Voltage      = 3.62 V
Transactions = No 
Interface    = SPI 
SPI overlap  = No 

Display driver = 9341
Display width  = 240 
Display height = 320 

MOSI    = D7 (GPIO 13)
MISO    = D6 (GPIO 12)
TFT_CS   = D0 (GPIO 16)
TFT_DC   = D8 (GPIO 15)
TFT_RST  = D1 (GPIO 5)

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.0 MHz 
[/code]

Any Ideas?

Bodmer, 2019-03-07:
Delete these lines:

#define TFT_MOSI 13
#define TFT_CLK  14
#define TFT_MISO 12

 as they are assigned automatically for the ESP8266.  The diagnostic report also shows that the SPI Clock pin has not been picked up as the "S" is missing in #define TFT_**S**CLK  14

marcvs, 2019-03-08:
Found the error: Apparently, the my settings (which swap CS and DC compared to the defaults)
#define TFT_CS   15 
#define TFT_DC   16
don't seem have any effect: They still show up as CS 16 and DC 15 (i.e. swapped) in the Read_User_Setup (good to have this example!).

Rewiring my setup to that it matches the defaults does the trick, but I still think this is a bug. Also: many pinouts indicate GPIO 15 to be CS.

Anyhow, thanks for the quick responses!!

Bodmer, 2019-03-09:
If the pins are not changing then this indicates you are not editing the correct file or more than one setup file is being called.

Make sure "User_Setup_Select.h" only calls up one setup file.

You can add a garbage line to the setup file you think the compiler is using to see if you get an error message such as:
```
zzz
```

closed 2019-03-09 by Bodmer

marcvs, 2019-03-11:
Yep, this is precisely what I did. I'm pretty sure I did edit the correct file, also because I think fixing the typo in TFT_CLK => TFT_SCLK did work. I'll re-confirm later this week.

marcvs, 2019-03-11:
Too bad I'm only doing this stuff after hours: I had defined the values in question in an earlier Session in User_Setup_Select.h, after User_Setup.h was included.

Many Sorries!

Bodmer, 2019-03-11:
No problem, thanks for letting me know. I am happy to help where I can.