Bodmer/TFT_eSPI #328: How to copy the specify area in the first sprite to the second sprite?

TridentTD, 2019-03-14:
I want to copy the area of point x1,y1  and point x2,y2 of the first sprite
to the new sprite.

How to copy this area of the sprite?

Bodmer, 2019-03-15:
You will need to write a function to do this. Use color = spr1.readPixel(x1,y1) and spr2.drawPixel(x2,y2,color) for each pixel you wish to copy. Using readPixel and drawPixel will handle all the coordinate bounds checking (prevents crashing the processor!) and color depths.

You should fing this gives reasonable performance. There are other ways of copying areas which are used in the scroll functions.

TridentTD, 2019-03-15:
When I scroll a text-sentence from right to left , the text by the scrolling function , at the first time must display all of text inner the sprite area. But I want to display a litle by litle of the sentence ,not by all of the sentence at the first time at the right side of the sprite.

Bodmer, 2019-03-15:
The Sprite_Scroll_16bit example shows scrolling text. It has a function called build_banner that you could adapt. Simply start the text with the xpos at the right side of the sprite and delete line 114.

TridentTD, 2019-03-18:
@Bodmer  Thank you for answer.

Do you mind to create   readRect() for  sprite  to read area from  sprite's memory 
not TFT_eSPI's readRect()   that read from the screen ?

I think it is helpful for copying one sprite to another sprite.

Thank you.

Bodmer, 2019-04-02:
I will add this to my long "to do" list, but it is not a priority.

In the meantime use readPixel and drawPixel to move the area you want between Sprites.

closed 2019-04-02 by Bodmer

MarekS57, 2026-09-08:
An old topic, but simple read/write.
Processing 5,000 pixels on the ESP32-S3 took 6 ms.
More control statements can be added.
Variables x and y refer to the source, and the destination always starts at _xd=0, _yd=0.

There is probably a more professional way to do this, but my skills are quite limited.

`
bool copyFragment(uint16_t x0, uint16_t y0, uint16_t x1, uint16_t y1, TFT_eSprite *sspr, TFT_eSprite *dspr ) {
// sspr source sprite
// dspr target sprite

   uint16_t _x0;
   uint16_t _y0;
   uint16_t _xd = 0;
   uint16_t _yd = 0;
   uint16_t _color;
   int16_t _width_dspr = dspr -> width(); 
   int16_t _height_dspr = dspr -> height();

  if ( _width_dspr < x1 || _height_dspr < y1) { 
    return false;
  }

 
  for (_y0 = y0; _y0 < y0 + y1; _y0++) {
    _xd = 0;
    for (_x0 = x0; _x0 < x0 + x1; _x0++) {
      _color = sspr -> readPixel( _x0, _y0);      
      dspr -> drawPixel( _xd, _yd, _color);       
      _xd++;
    }
    _yd++;
  }

  return true;
}
`