Bodmer/TFT_eSPI #329: Overwriting text on the display

pierpe83, 2019-03-17:
Hi
I have a problem with displaying the variable on the display.
When the variable value changes, it is overwritten to the old value and it becomes cluttered on the display.
I did not find any text cleaning function on the display.
How to deal with this?

VikJr, 2019-03-17:
> tft.setTextSize(3); tft.setTextColor(TFT_RED); tft.setCursor(0, 10); tft.println("Temperature"); tft.setTextColor(TFT_YELLOW); tft.setTextSize(5); tft.setCursor(0, 40); tft.println(pinData); tft.setCursor(60, 32); tft.println("o"); tft.setCursor(88, 40); tft.println("C");
--------------------

	pierpe83 Try this option -
 tft.setTextSize(3); tft.setTextColor(TFT_RED,TFT_BLACK); tft.setCursor(0, 10); tft.println("Temperature"); tft.setTextColor(TFT_YELLOW,TFT_BLACK); tft.setTextSize(5); tft.setCursor(0, 40); tft.println(pinData); tft.setCursor(60, 32); tft.println("o"); tft.setCursor(88, 40); tft.println("C");

---------------------
 tft.setTextColor(TFT_RED,TFT_BLACK)
TFT_RED - text color is red.
TFT_BLACK - background is black.

Bodmer, 2019-03-19:
As VikJr mentions you can specify a background colour, however as the digit count changes part of the old longer bumber can be left behind. One way around this is to add padding spaces.

drawNumber(), drawFloat() and drawString() functions can all be used with automatic padding to a defined width and with a defined datum.  See examples:

TFT_String_Align and TFT_Padding_Demo

holla2040, 2019-03-26:
Bodmer, thanks the padding example reference. setTextPadding works great for me.

closed 2019-04-01 by Bodmer