Bodmer/TFT_eSPI #330: Support for HX8347D ?

68gt500, 2019-03-19:
Trying to get my older MCUFriend shield to work with an WEMOS ESP32 Uno type board.

Did the hw mods - David´s mcufriend_kbv works fine - the diag  says it is a HX8347D - code 0x4747

your read_user_setup says:
[code]
TFT_eSPI ver = 1.4.5
Processor    = ESP32
Frequency    = 240 MHz
Transactions = Yes 
Interface    = Parallel 
Display driver = 9341
Display width  = 240 
Display height = 320 

TFT_CS   = D33 (GPIO 33)
TFT_DC   = D15 (GPIO 15)
TFT_RST  = D32 (GPIO 32)

TFT_WR   = D4 (GPIO 4)
TFT_RD   = D2 (GPIO 2)

TFT_D0   = D12 (GPIO 12)
TFT_D1   = D13 (GPIO 13)
TFT_D2   = D26 (GPIO 26)
TFT_D3   = D25 (GPIO 25)
TFT_D4   = D17 (GPIO 17)
TFT_D5   = D16 (GPIO 16)
TFT_D6   = D27 (GPIO 27)
TFT_D7   = D14 (GPIO 14)

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

[/code]

Any chance to get it working ?
Thanks

Bodmer, 2019-03-19:
Edit the library User_Setup.h file to select the display you have.

68gt500, 2019-03-20:
I could not find any  entry for HX8347D  .. That is why I asked ;-)

#define ILI9341_DRIVER
//#define ST7735_DRIVER      // Define additional parameters below for this display
//#define ILI9163_DRIVER     // Define additional parameters below for this display
//#define S6D02A1_DRIVER
//#define RPI_ILI9486_DRIVER // 20MHz maximum SPI
//#define HX8357D_DRIVER
//#define ILI9481_DRIVER
//#define ILI9486_DRIVER
//#define ILI9488_DRIVER     // WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO does NOT tristate when CS is high)
//#define ST7789_DRIVER      // Define additional parameters below for this display
//#define R61581_DRIVER

Bodmer, 2019-03-20:
Unfortunately that display controller is not supported at the moment.

68gt500, 2019-03-20:
OK, I just bought an ILI9481 Shield, should be here by mid April. 
 Really looking forward to have that weather station running in full screen mode.
Sure looks fantastic.

closed 2019-03-20 by 68gt500

Bodmer, 2019-03-20:
Note that the DarkSky weather library example fills a 320x240 screen and thus will only occupy half a 320x480 screen.

adolfonovo, 2021-08-30:
Hi @Bodmer ... is there any chance to include this driver?... 
Thanks and congrats for your job