Bodmer/TFT_eSPI #332: ESP32-TTGO+TFT-ILI9341

MorisB, 2019-03-22:
I edit User_Setup.h file with this lines

#define ILI9341_DRIVER

// Display to ESP32 pin connections
// Connect TFT VCC and LED pins to ESP32 5V
// Connect TFT GND pin to ESP32 0V
#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS    5
#define TFT_DC    0
#define TFT_RST   -1  // connect Direct to RST

#define TOUCH_CS 13

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF

#define SMOOTH_FONT

#define SPI_FREQUENCY  40000000

#define SPI_TOUCH_FREQUENCY  2500000

The graphics wort fix,
But in the example  Keypad_240x320.ino
I set Rotation like this    tft.setRotation(2);
And can't calibrate, 

I should connect the other touch pins on TFT, to ESP32 board?

Bodmer, 2019-03-23:
Hi,

Yes, you need to connect the touch pins.  In your setup the extra connections would be:

T_IRQ - no connection
T_DO connect to ESP32 pin 19 (MISO)
T_DIN connect to ESP32 pin 23 (MOSI)
T_CS connect to ESP32 pin 13
T_CLK connect to ESP32 pin 18 (SCLK)

Run the Test_Touch_Controller example to test the touch controller interface.

Make sure your display actually has a touch screen film fitted and the XPT2046 controller chip fited as well.  Note that some displays sold have the XPT2046 fitted but do not have the touch sensitive film fitted to the screen.

MorisB, 2019-03-23:
thanks for the reply,
I've included photos of the chip and the film, I hope it's all right, right?
Do I need to add the connections you recorded?
Or do I have to remove the connections that exist for the display, and instead connect them to the touch pins?
[image]
[image]

Bodmer, 2019-03-23:
The screen you have looks good.

Yes you need to add wires to connect the ESP32 to the touch controller on the TFT, leave the existing connections to the TFT otherwise the screen will not work!

No additions to the User_Setup.h file are needed.

MorisB, 2019-03-23:
I'm sorry, I do not seem to understand.

According to your recommendation at first to connect like that.
#defineTFT_MISO 19
#defineTFT_MOSI 23
#defineTFT_SCLK 18
#defineTFT_CS5
#define TFT_DC 0

If I leave these connections, then connect them as well?
T_IRQ - no connection
T_DO connect to ESP32 pin 19 (MISO)
T_DIN connect to ESP32 pin 23 (MOSI)
T_CS connect to ESP32 pin 13
T_CLK connect to ESP32 pin 18 (SCLK)

This means that the pin from ESP32-19 connects to two places,
TFT_MISO 19

And also to
T_DO connect to ESP32 pin 19 (MISO)?

And so with the other pins?

Bodmer, 2019-03-24:
Correct, that is how an SPI bus works, it can communicate with multiple devices on the same MOSI,MISO and SCLK pins. Connect as instructed.

MorisB, 2019-03-24:
OK, Thanks, it's work, but something not fine.

I run Touch_Controller.ino Example
And I get this lines in serial monitor.

  uint16_t calData[5] = { 639, 2934, 291, 2555, 7 };
  tft.setTouch(calData);

I put this lines in the example.
I see dots on display, but not in same place I touch.

It's look like the calibrate not work.
also the On_Off_Button.ino example, not reactive to the touches.

Where am I wrong?

Bodmer, 2019-03-24:
Run the Touch_calibrate example. Do not change the sketch. Do the white touch dots appear in the right place?

MorisB, 2019-03-24:
Bodmer,

First of all,
Thanks for your help, and I apologize for my English.

I run the example, but I have to press the calibration, at the end of the arrow, not at its tip.
I thing I have a problem with the display, and I do not know if it's because of the sketch.
When I manage to see white dots, they are not where I click.

If I press the arrow in the corners, in some of the arrows, I suddenly get a white screen.
I have to push the RST button on ESP32, to start the sketch.

Could this be the sketch? Or is it a screen calibrating?

Bodmer, 2019-03-24:
The screen should not go white during calibration, perhaps you are disturbing the wires connecting to the screen?

If you press positions A B C D and E in the diagram below at what letter position do the dots appear?

___________________________________
|                                                    |
|         A                            B           |
|                                                    |
|                          C                        |
|                                                    |
|         D                            E           |
|___________________________________|

MorisB, 2019-03-25:
If it does not get stuck, with a white screen.
I get what's in the picture attached.

I have to press several times until the point is displayed,
And it is located in various places, scattered around point A for example.

[image]

MorisB, 2019-03-28:
Do you have an idea how to solve the problem?

Thanks.

Bodmer, 2019-04-01:
You seem to have a display that is generating odd values.

A white screen indicates the ESP32 is crashing which is odd. Make sure your USB power supply in providing enough power.

I have run out of ideas to help unfortunately.

MorisB, 2019-04-02:
OK, thanks for the response.
Do you have a solution to this problem?
" You seem to have a display that is generating odd values. "

Another thing about the BMP/JPG.
If I want to display JPG images from a SD card connected to TTGO,
What are the connections I have to make, and what example should I use?

Thanks a lot

Bodmer, 2019-04-08:
Sorry, I do not have a solution as my screens seem to work fine.
[
This example ](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Generic/ESP32_SDcard_jpeg )would be a good starting point for jpegs from SD card.

closed 2019-04-08 by Bodmer

MorisB, 2019-04-11:
OK Thanks.