Bodmer/TFT_eSPI #333: ILI9327 support

onurguc, 2019-03-24:
Hi,

Any plans to support ILI9327 display?  The display has minimal support for arduino but there is simply nothing for ESP32.  I have the ILI9327 shield.  I believe it is an 8 bit unit.

Thanks,
-Onur

Bodmer, 2019-03-24:
Try this setup:

```

#define ESP32_PARALLEL

#define ILI9486_DRIVER

// ESP32 pins used for UNO format board
#define TFT_CS   33  // Chip select control pin
#define TFT_DC   15  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST  32  // Reset pin

#define TFT_WR    4  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD    2

#define TFT_D0   12  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits
#define TFT_D2   26
#define TFT_D3   25
#define TFT_D4   17
#define TFT_D5   16
#define TFT_D6   27
#define TFT_D7   14

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT
```

and in the sketch define the screen size thus:

TFT_eSPI tft = TFT_eSPI(240,400);

onurguc, 2019-03-25:
I did try it with D1 R32 ESP32 uno shield. I did the hardware modification described too.  My TFT is this one https://tr.aliexpress.com/item/3-2-TFT-LCD-Display-module-Touch-Screen-Shield-board-onboard-temperature-sensor-w-Touch-Pen/32755473754.html?spm=a2g0s.9042311.0.0.27424c4dfbQ4F9
and the ESP32 is this one https://www.aliexpress.com/snapshot/0.html?spm=a2g0s.9042311.0.0.27424c4dY1p816&orderId=98328956134973&productId=32916569687

I must be doing something.  I just suspect the ILI9486 more pins than ILIU9327.

Bodmer, 2019-04-01:
Your board does not have the "standard£ pinout for these boards.

The example setup assumes you have a board with the pinout here:

https://os.mbed.com/users/rlanghbv/code/mcufriend_24_tft_lcd/wiki/Hardware

onurguc, 2019-04-04:
Well I decided to use the 9327 in other project.  But I managed to work ILI9486 display with no problems.  I do have a question though.   If I don't need the touch functionality, how can I disable touch functions and free all the digital pins for other sensors?  Is there a quick way to accomplish this?

Bodmer, 2019-04-08:
If you do not define a touch pin in the setup file then the touch functions will not be loaded.

The touch pins are dual purpose and thus cannot be freed for other functions.

closed 2019-04-08 by Bodmer