Bodmer/TFT_eSPI #335: ESP32 TFT ILI9481 Mirrored output

68gt500, 2019-03-28:
Just got my cheap chinese ILI9481 TFT UNO shield.
Works - kind of - but the screen output is mirrored 

[image]
Really nice colors and good resolution, not bad for the money.
Dont´t mind the slash, just on the protective foil

Any Ideas ?

68gt500, 2019-03-28:
A few additional findings:
Demos:
Font1 -- mirror
Font2,3,4 -- ok
Unicode Test -- nothing, white screen
Local Custom Font -- ok
TFT Padding -- ok
TFT Rainbow - mirror -- see above
Free Font Demo -- ok
3D Cube OK

Dark Sky - all mirrored :-(

[image]

68gt500, 2019-03-28:
Sorry, hit wrong button

68gt500, 2019-03-29:
Found out, that using a ILI9486 driver instead of the ILI9481 the mirrored problem is solved.
user_setup.h:
//#define ILI9481_DRIVER  //mirrored
#define ILI9486_DRIVER   //okay

josephsr, 2019-04-01:
Can you send me link for the display that you are using? We have a similar display (9481) but unable to get it to work

68gt500, 2019-04-01:
Suer: https://www.ebay.de/itm/3-5inch-TFT-color-screen-module-320X480-ultra-HD-support-for-Arduino-for-UNO-R/263984004356?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m2749.l2649
This is the display, it claims: IC:ILI9481
Hookup was very easy:  Do the described mods to the board and just plug it in. All pin assignments stay standard.

Bodmer, 2019-04-01:
@68gt500 Can you post a picture of the back (pcb side) of your display please.

68gt500, 2019-04-02:
This is a link to the ebay picture: https://i.ebayimg.com/images/g/2ScAAOSwr9VZ1~yz/s-l1600.jpg

I will check tonight, if it is exactly like the one I received.

68gt500, 2019-04-02:
[image]
Here is a pic of my board, which is not exactly as the image from ebay.

Bodmer, 2019-04-02:
That looks like an ILI9486 board so you have selected the right driver.

It is not unusual for averts from some low cost vendors to be wrong which is not very helpful to buyer!

closed 2019-04-02 by Bodmer

fuxifex, 2019-09-26:
Hi, can you please provide a Wiring Diagram? (I'm using a ESP32 DevKit-C) 
Thanks!