Bodmer/TFT_eSPI #339: tft.init() makes pins unsuable

firepower2k11, 2019-04-06:
I'm using the following setup:

- LoLin NodeMCU v3
- ST7789 1.3" IPS (SPI without CS pin)
- Serial sensor

User Setup config:
#define TFT_DC   PIN_D3
#define TFT_RST  PIN_D4
#define TFT_MOSI PIN_D7
(no more pin definitions but the default other definitions like driver, fonts, ...)
(CLK is on PIN_D5, although not defined)

After the tft.init() function I can't use the software serial function (in this case named SerialCommunication.begin(BAUD)) anymore.
If I use the overlap mode (#define TFT_SPI_OVERLAP) the sensor works fine, but I can't go this way since my display has no CS pin and therefore I can't use the hardware SPI pins of the board.

So the question is:
How can I stop tft.init() breaking other objects pin usage?
After trying a bit at least D0, D6 and D8 can't be used anymore.

Example pseudocode how it doesn't work:
```
TFT_eSPI tft = TFT_eSPI();
SoftwareSerial SerialCom (RX,TX);

void setup{
    SerialCommunication.begin(BAUD));
    tft.init();
}
void loop{
    SerialCommunication.read(data);
    Serial.println(data); ===============>> Returns nothing
    tft.println(data); ===============>> Returns nothing
}
```

Pseudocode how it works but only until tft.init()
```
TFT_eSPI tft = TFT_eSPI();
SoftwareSerial SerialCom (RX,TX);

void setup{
    SerialCommunication.begin(BAUD));
    SerialCommunication.read(data);
    Serial.println(data); ===============>> Returns data
    tft.init();
}
void loop{
    SerialCommunication.read(data);
    Serial.println(data); ===============>> Returns nothing
    tft.println(data); ===============>> Returns nothing
}
```

Bodmer, 2019-04-08:
D5, D6, D7 and D8 are [allocated to the SPI bus](https://www.electronicwings.com/nodemcu/nodemcu-spi-with-arduino-ide).

If you do not define the SPI CS then the library assumes you are using D8.  If you define TFT_CS as -1, then it should free up D8.

Depending on which software serial library you are using you may find GPIO 16 (D0) is not usable whether you are using SPI or not.

closed 2019-04-08 by Bodmer

firepower2k11, 2019-04-08:
I didn't use the HW SPI pin layout since I had another I²C sensor library which had D5 / D6 somehow hardcoded in it.
Now I've changed the display to D5-D8 and it seems to work now.
Software serial on D0/D1 also works for me now.

Thank you very much!