Bodmer/TFT_eSPI #340: ESP32_SDcard_jpeg

amirjak, 2019-04-09:
[image]

Bodmer, 2019-04-12:
How have you wired the ESP32 to the shield?

Do the TFT_eSPI library examples display correctly?

amirjak, 2019-04-12:
hi
thank for answering
How have you wired the ESP32 to the shield?
in file **User_Setup.h**
#define ILI9486_DRIVER
#define ESP32_PARALLEL

// The ESP32 and TFT the pins used for testing are:
#define TFT_CS   33  // Chip select control pin (library pulls permanently low
#define TFT_DC   15  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST  32  // Reset pin, toggles on startup

#define TFT_WR    4  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD    2  // Read strobe control pin

#define TFT_D0   12  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits.
#define TFT_D2   26  // Pins can be randomly assigned, this does not affect
#define TFT_D3   25  // TFT screen update performance.
#define TFT_D4   17
#define TFT_D5   16
#define TFT_D6   27
#define TFT_D7   14
Do the TFT_eSPI library examples display correctly? **YES**

[image]
with sd card module connect to DOIT Esp32 DevKit v1 work correctly
_______SD_SS ________________D5____cs
______SD_DI _________________D19____MISO
______SD_DO________________D23 ___MOSI
______SD_SCK_______________D18 ____CLK

But with  sd card on tft module  not working
i wired like this
_______SD_SS ________________D5____cs
______SD_DI _________________D19____MISO
______SD_DO________________D23 ___MOSI
______SD_SCK_______________D18 ____CLK

Bodmer, 2019-04-12:
If you only connect power (+5V and 0V) and SPI lines to the TFT shield SD card pins does the SD Card interface work?

pipi61, 2019-04-12:
Hi!
This sdcard module contain power regulator, and level converter.
Try another simple sdcard module without regulator and without converter

amirjak, 2019-04-13:
> If you only connect power (+5V and 0V) and SPI lines to the TFT shield SD card pins does the SD Card interface work?

NOT WORKING

Bodmer, 2019-04-13:
It looks like a hardware fault or an incompatibilty for some reason.

J1-4 are simply reference names for the header strips so youcan identify their position and reference the schematic.

I will close as the problem you have does not appear tobe due to the library.

closed 2019-04-13 by Bodmer