Bodmer/TFT_eSPI #341: LCD and PMS7003 Sensor

Tokn59, 2019-04-10:
Hello, I am using your library (BIG THANKS !) But I run into a strange issue. I am using a sketch wicxh reads data from a PMS7003 sensor (serial over pins 6 and 8 on ESP8266. The LCD is connected as follows :

/*  SPI:                                                */
/*      - ILI9341   SCK     => GPIO14                   */
/*      - ILI9341   MISO    NC                          */
/*      - ILI9341   MOSI    => GPIO13                   */
/*      - ILI9341   D/C     => GPIO16                   */
/*      - ILI9341   RST     => RST                      */
/*      - ILI9341   CS      => GND                      */
/*      - ILI9341   LED     => 3V3 */

As soon as I put the LCD.begin() code in my sketch (full sketch available on request) the sensor gets read only once and then no furter data is retrieved from the sensor. When I remove the LCD.begin() statement everything works fine. Any clues ?

Many thanks & kind regards,

Tony

Bodmer, 2019-04-11:
Pins D6 and D8 are used for the SPI bus so there is a conflict and you need to use different pins for the sensor.

This is the Arduino Dx pin to GPIO mapping:
```
// These are the pins for all ESP8266 boards
//      Name   GPIO    Function
#define PIN_D0  16  // WAKE
#define PIN_D1   5  // User purpose
#define PIN_D2   4  // User purpose
#define PIN_D3   0  // Low on boot means enter FLASH mode
#define PIN_D4   2  // TXD1 (must be high on boot to go to UART0 FLASH mode)
#define PIN_D5  14  // HSCLK
#define PIN_D6  12  // HMISO
#define PIN_D7  13  // HMOSI  RXD2
#define PIN_D8  15  // HCS    TXD0 (must be low on boot to enter UART0 FLASH mode)
#define PIN_D9   3  //        RXD0
#define PIN_D10  1  //        TXD0

#define PIN_MOSI 8  // SD1
#define PIN_MISO 7  // SD0
#define PIN_SCLK 6  // CLK
#define PIN_HWCS 0  // D3

#define PIN_D11  9  // SD2
#define PIN_D12 10  // SD4

```

Bodmer, 2019-04-15:
I am closing this as I think this is not an issue with the library but if I am wrong then I can re-open the issue.

closed 2019-04-15 by Bodmer