Bodmer/TFT_eSPI #344: RA8875 support for 800x480 5inch, 7inch displays

knopserl, 2019-04-16:


knopserl, 2019-04-16:
First I want to thanks you for this geat library. I really like it and it's much faster than other libs I have tested.
We are now looking for a higher resolution SPI TFT Touch display with 800x480 and found some with the RA8875 controller. Is there any chance to get support for such a controller. I found other libs with that chip but wanted to have one source with this library and this performance and touch support rather than another project for another driver library.
Here is such a display we might want to use
https://www.buydisplay.com/default/5-inch-tft-lcd-module-800x480-display-controller-i2c-serial-spi

Bodmer, 2019-04-20:
I have no plans to support that controller.  One day I may buy a screen and adapt the library but at the moment I have no projects that will need a large display.

closed 2019-04-20 by Bodmer

knopserl, 2019-05-05:
@Bodmer thanks at least for the quick answer. What a pitty, I'm doing quite a lot of projects with displays and the latest (railroad control) requires a higher resolution with 800x480 and 5" (or 7"). There are very good displays with RA8875 and SPI which are very powerfull and ultrafast. Sine I like your library, it would be great to have this very popular chip supported. Otehrwise I have to switch to another lib and rewrite my code. BTW: There is an Aadafruit lib available for RA8875, so there is at least some code which might be reusable for this lib, but unfortunatly I'm not capabale to include it to this library by meyself