Bodmer/TFT_eSPI #345: 8b Parallel + 4-wire Touch

ImpulseAdventure, 2019-04-17:
Hi @Bodmer -- Thanks again for the great work on TFT_eSPI!

I was wondering if you might be able to briefly comment on the reason for the `TFT_CS` "permanent low" approach in `ESP32_PARALLEL` mode?

The reason for my question is that a chip select force may conflict with the use of certain pin-sharing displays that depend on an analog 4-wire touch interface.

thanks

Bodmer, 2019-04-20:
TFT_CS is held low for performance reasons and the library does not support touch control on these UNO shields, extra wires would be needed to make LCD_WR and LCD_RS analogue pins. If you do wish to read the touch screen then you will need to drive the TFT_CS line high in your sketch while reading the screen analogue readings.

closed 2019-04-20 by Bodmer

ImpulseAdventure, 2019-04-23:
Thank you @Bodmer for the confirmation. I had already wrapped GUIslice's touch driver with CS toggles (and presumed it may have been an optimization for TFT_eSPI's very high display throughput), but thought I'd double-check in case there were other issues to watch out for. thx