Bodmer/TFT_eSPI #348: Compilation error with latest esp8266 GIT

luc-github, 2019-04-22:
First - thank you for your great library 
with latest esp8266 core GIT compilation failed with sample like `UTFT_demo.ino`, no issue on ESP32.
It seems due to this recent commit : 
https://github.com/esp8266/Arduino/commit/885276e75cbad9aa838217edbcdbac7aa5725719
It is out of my knowledge to know why they did this but may be the library need some update to handle this.

```
Compiling library "TFT_eSPI"
"C:\\Users\\user\\Documents\\Arduino\\hardware\\esp8266com\\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\user\\Documents\\Arduino\\hardware\\esp8266com\\esp8266/tools/sdk/include" "-IC:\\Users\\user\\Documents\\Arduino\\hardware\\esp8266com\\esp8266/tools/sdk/lwip2/include" "-IC:\\Users\\user\\Documents\\Arduino\\hardware\\esp8266com\\esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\user\\AppData\\Local\\Temp\\arduino_build_134255/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -fno-exceptions -DNONOSDK221=1 -DF_CPU=160000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10809 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=2 -DFLASHMODE_DIO -DESP8266 "-IC:\\Users\\user\\Documents\\Arduino\\hardware\\esp8266com\\esp8266\\cores\\esp8266" "-IC:\\Users\\user\\Documents\\Arduino\\hardware\\esp8266com\\esp8266\\variants\\generic" "-IC:\\Users\\user\\Documents\\Arduino\\hardware\\esp8266com\\esp8266\\libraries\\SPI" "-IC:\\Users\\user\\Documents\\Arduino\\libraries\\TFT_eSPI" "C:\\Users\\user\\Documents\\Arduino\\libraries\\TFT_eSPI\\TFT_eSPI.cpp" -o "C:\\Users\\user\\AppData\\Local\\Temp\\arduino_build_134255\\libraries\\TFT_eSPI\\TFT_eSPI.cpp.o"
In file included from C:\Users\user\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/pgmspace.h:7:0,

                 from C:\Users\user\Documents\Arduino\libraries\TFT_eSPI/Fonts/glcdfont.c:6,

                 from C:\Users\user\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.h:62,

                 from C:\Users\user\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:17:

C:\Users\user\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'virtual int16_t TFT_eSPI::drawChar(uint16_t, int32_t, int32_t, uint8_t)':

C:\Users\user\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:4273:84: error: invalid conversion from 'uint32_t {aka unsigned int}' to 'const void*' [-fpermissive]

       flash_address = pgm_read_dword( pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *) );

                                                                                    ^

C:\Users\user\Documents\Arduino\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:114:57: note: in definition of macro 'pgm_read_dword'

     #define pgm_read_dword(a)  pgm_read_dword_unaligned(a)

                                                         ^

C:\Users\user\Documents\Arduino\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:85:72: error:   initializing argument 1 of 'uint32_t pgm_read_dword_unaligned(const void*)' [-fpermissive]

 __attribute__((optimize("-O3"), always_inline)) static inline uint32_t pgm_read_dword_unaligned(const void *addr) {

                                                                        ^

C:\Users\user\Documents\Arduino\libraries\TFT_eSPI\Extensions/Sprite.cpp: In member function 'virtual int16_t TFT_eSprite::drawChar(uint16_t, int32_t, int32_t, uint8_t)':

C:\Users\user\Documents\Arduino\libraries\TFT_eSPI\Extensions/Sprite.cpp:1682:84: error: invalid conversion from 'uint32_t {aka unsigned int}' to 'const void*' [-fpermissive]

       flash_address = pgm_read_dword( pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *) );

                                                                                    ^

C:\Users\user\Documents\Arduino\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:114:57: note: in definition of macro 'pgm_read_dword'

     #define pgm_read_dword(a)  pgm_read_dword_unaligned(a)

                                                         ^

C:\Users\user\Documents\Arduino\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:85:72: error:   initializing argument 1 of 'uint32_t pgm_read_dword_unaligned(const void*)' [-fpermissive]

 __attribute__((optimize("-O3"), always_inline)) static inline uint32_t pgm_read_dword_unaligned(const void *addr) {

                                                                        ^
```

Bodmer, 2019-04-28:
I have only tried the release versions of the board package downloaded via the Arduino boards manager and these work OK.

I avoid copies direct from the latest library development tweaks on Github as they tend to be buggy.

This appears to be down to type casting of pointer types ans is likely to break other libraries.

I will keep an eye on the situation and see if the problem propagates to a realese copy.

luc-github, 2019-04-28:
Yes, it was more a head up as rollback the change I mentioned solve the issue currently, but this change was done on purpose, so if no one raise it will be in release, and I am not skilled to discuss why they do this change
Thank you again for your great work

Bodmer, 2019-05-01:
OK, I will leave this open for now and see if the issue migrates to the ESP8266 board package.

elgerg, 2019-05-11:
Hi Bodmer,

Yes it looks like 2.5.1 suffers:

[Starting] Verify sketch - ESP8266-12_SofpApSetup_V3.2.ino
Please see the build logs in Output path: C:\Temp\Arduino\__BuildArea\SoftAPSetupV3
Loading configuration...
Initialising packages...
Preparing boards...
Verifying...
Build options changed, rebuilding all
In file included from C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266/pgmspace.h:7:0,

                 from C:\Users\user\Documents\Arduino\libraries\TFT_eSPI/Fonts/glcdfont.c:6,

                 from C:\Users\user\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.h:62,

                 from C:\Users\user\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:17:

C:\Users\user\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'virtual int16_t TFT_eSPI::drawChar(uint16_t, int32_t, int32_t, uint8_t)':

C:\Users\user\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:4262:84: error: invalid conversion from 'uint32_t {aka unsigned int}' to 'const void*' [-fpermissive]

       flash_address = pgm_read_dword( pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *) );

                                                                                    ^

C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:114:57: note: in definition of macro 'pgm_read_dword'

     #define pgm_read_dword(a)  pgm_read_dword_unaligned(a)

                                                         ^

C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:85:72: error:   initializing argument 1 of 'uint32_t pgm_read_dword_unaligned(const void*)' [-fpermissive]

 __attribute__((optimize("-O3"), always_inline)) static inline uint32_t pgm_read_dword_unaligned(const void *addr) {

                                                                        ^

C:\Users\user\Documents\Arduino\libraries\TFT_eSPI\Extensions/Sprite.cpp: In member function 'virtual int16_t TFT_eSprite::drawChar(uint16_t, int32_t, int32_t, uint8_t)':

C:\Users\user\Documents\Arduino\libraries\TFT_eSPI\Extensions/Sprite.cpp:1682:84: error: invalid conversion from 'uint32_t {aka unsigned int}' to 'const void*' [-fpermissive]

       flash_address = pgm_read_dword( pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *) );

                                                                                    ^

C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:114:57: note: in definition of macro 'pgm_read_dword'

     #define pgm_read_dword(a)  pgm_read_dword_unaligned(a)

                                                         ^

C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:85:72: error:   initializing argument 1 of 'uint32_t pgm_read_dword_unaligned(const void*)' [-fpermissive]

 __attribute__((optimize("-O3"), always_inline)) static inline uint32_t pgm_read_dword_unaligned(const void *addr) {

                                                                        ^

exit status 1
[Error] Exit with code=1

Bodmer, 2019-05-19:
see #361

closed 2019-05-19 by Bodmer

luc-github, 2019-05-20:
Sorry for the delay but I am currently travelling, so no much free time t test.
Hi I am sorry to say the sample `UTFT_demo.ino` still cannot compile :
I have downloaded latest TFT_eSPI today and use a clean installation of esp core 2.5.1 - I do not changed anything in User_Setup.h as it is for esp8266 and ILI9341
and I get :
```
Compiling library "TFT_eSPI"
"C:\\Users\\User\\Documents\\ArduinoData\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\User\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.1/tools/sdk/include" "-IC:\\Users\\User\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.1/tools/sdk/lwip2/include" "-IC:\\Users\\User\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.1/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\User\\AppData\\Local\\Temp\\arduino_build_970810/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -fno-exceptions -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10809 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=2 -DFLASHMODE_DOUT -DESP8266 "-IC:\\Users\\User\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.1\\cores\\esp8266" "-IC:\\Users\\User\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.1\\variants\\generic" "-IC:\\Users\\User\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.1\\libraries\\SPI" "-IC:\\Users\\User\\Documents\\Arduino\\libraries\\TFT_eSPI" "C:\\Users\\User\\Documents\\Arduino\\libraries\\TFT_eSPI\\TFT_eSPI.cpp" -o "C:\\Users\\User\\AppData\\Local\\Temp\\arduino_build_970810\\libraries\\TFT_eSPI\\TFT_eSPI.cpp.o"
In file included from C:\Users\User\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266/pgmspace.h:7:0,

                 from C:\Users\User\Documents\Arduino\libraries\TFT_eSPI/Fonts/glcdfont.c:6,

                 from C:\Users\User\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.h:62,

                 from C:\Users\User\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:17:

C:\Users\User\Documents\Arduino\libraries\TFT_eSPI\Extensions/Sprite.cpp: In member function 'virtual int16_t TFT_eSprite::drawChar(uint16_t, int32_t, int32_t, uint8_t)':

C:\Users\User\Documents\Arduino\libraries\TFT_eSPI\Extensions/Sprite.cpp:1682:84: error: invalid conversion from 'uint32_t {aka unsigned int}' to 'const void*' [-fpermissive]

       flash_address = pgm_read_dword( pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *) );

                                                                                    ^

C:\Users\User\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.1/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:114:57: note: in definition of macro 'pgm_read_dword'

     #define pgm_read_dword(a)  pgm_read_dword_unaligned(a)

                                                         ^

C:\Users\User\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.1/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:85:72: error:   initializing argument 1 of 'uint32_t pgm_read_dword_unaligned(const void*)' [-fpermissive]

 __attribute__((optimize("-O3"), always_inline)) static inline uint32_t pgm_read_dword_unaligned(const void *addr) {

                                                                        ^

Using library SPI at version 1.0 in folder: C:\Users\User\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.1\libraries\SPI 
Using library TFT_eSPI at version 1.4.10 in folder: C:\Users\User\Documents\Arduino\libraries\TFT_eSPI 
exit status 1
Error compiling for board Generic ESP8266 Module.
```

looks like pgm issue is still there - should I open new ticket ? or can you reopen this ticket as it is not actually solved ?

Bodmer, 2019-05-20:
Thanks, have updated Sprite.cpp, I forgot that one!