Bodmer/TFT_eSPI #349: Can't use TFT 3.5 inch Uno Shield with Esp32 when use 8bit parallel.

thesofial, 2019-04-23:
Hi sir, I used 3.5 FTF Shield for Esp32, use 8 bit parallel to tranfer data, but it not working. If I use TFT with Mega or Uno, it's work great. I don't know what is issue, please help me. This is the code.
```
#define ESP32_PARALLEL
#define ILI9341_DRIVER
#include <TFT_eSPI.h> // Hardware-specific library
#include <SPI.h>

// The ESP32 and TFT the pins used for testing are:
#define TFT_CS   33  // Chip select control pin (library pulls permanently low
#define TFT_DC   15  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST  32  // Reset pin, toggles on startup
#define TFT_WR    4  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD    2  // Read strobe control pin

#define TFT_D0   12  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits.
#define TFT_D2   26  // Pins can be randomly assigned, this does not affect
#define TFT_D3   25  // TFT screen update performance.
#define TFT_D4   17
#define TFT_D5   16
#define TFT_D6   27
#define TFT_D7   14
TFT_eSPI tft;

```
Thank for your help, sir !

thesofial, 2019-04-23:
[image]

I use this TFT

Bodmer, 2019-05-01:
I cannot see anything wrong with your pin choice. I assume you have edited the User_Setup.h file in the library? Your post suggests you are trying to include the setup in your sketch!

Run the "Read_User_Setup" example and look at the serial monitor output to see if the compiler is picking up your settings.

It is best to use an UNO format ESP32 as you can get ringing on the high speed signals and ground bounce which leads to spurious signals due to the high edge speeds.

Unfortunately I am unable to help further.

Bodmer, 2019-06-11:
Closed pending feedback.

closed 2019-06-11 by Bodmer