Bodmer/TFT_eSPI #350: Need a degree sign (°)

quorle, 2019-04-24:
Hello, I currently use their lib, but if I enter the following for example:

tft.setCursor(145, 50 ,2); tft.setTextColor(TFT_WHITE);  tft.setTextSize(2); tft.println("°C");

 then I get the degree symbol is not displayed. How do I get it displayed the easiest?

elgerg, 2019-04-24:
Try:

tft.print("`");

quorle, 2019-04-24:
> tft.print("`");

Unfortunately that does not bring the desired sign. The desired character would be: ° C -> degrees centigrade.

Also, I have the problem, as soon as a number changes, it is simply replaced by the new one. Example: the number 6 is on the display and is replaced by 7, now both numbers are in the display.

elgerg, 2019-04-24:
What sign does it give you?

There is a specific change that makes that symbol represent a degree symbol.

Ive used it a lot and it works as the attached.

When you say one number is displayed by another, are you clearing the display/section first?
[image]

quorle, 2019-04-24:
Ok, so in my case only "` "will be shown on the display. With you, it looks like it should look like me. Here is a part of my code that I use:

```
void display() {
  // Fill screen with grey so we can see the effect of printing with and without
  // a background colour defined
 // tft.fillScreen(TFT_BLACK);

  // Set "cursor" at top left corner of display (0,0) and select font 2
  // (cursor will move to next line automatically during printing with 'tft.println'
  //  or stay on the line is there is room for the text with tft.print)
 // tft.setCursor(0, 0, 2);
  // Set the font colour to be white with a black background, set text size multiplier to 1
//  tft.fillScreen(TFT_BLACK); tft.setCursor(0, 0 ,2); tft.setTextColor(TFT_WHITE,TFT_BLACK);  tft.setTextSize(1); tft.println("Hello World!");
  // We can now plot text on screen using the "print" class
  tft.setCursor(65, 50 ,2); tft.setTextColor(TFT_WHITE);  tft.setTextSize(2); tft.println(Temperatur_IST); // Temperatur aus Variable auf das Display schreiben
  tft.setCursor(145, 50 ,2); tft.setTextColor(TFT_WHITE);  tft.setTextSize(2); tft.print("°C");  //tft.print("`");        // Grad Celsius auf das Display schreiben
  tft.setCursor(65, 80 ,2); tft.setTextColor(TFT_WHITE);  tft.setTextSize(2); tft.println(Feuchte);        // Feuchte aus Variable auf das Display schreiben
  tft.setCursor(145, 80 ,2); tft.setTextColor(TFT_WHITE);  tft.setTextSize(2); tft.println("%");           // % auf das Display schreiben

   //tft.pushImage(1, 60, PlusWidth, PlusHeight, Plus); // Icon auf Bildschirm darstellen
   //tft.pushImage(185, 60, MinusWidth, MinusHeight, Minus); // Icon auf Bildschirm darstellen
}
```

Sorry, for the german writing, but I'm from Germany

elgerg, 2019-04-25:
Try this:

`
void display() {
  // Fill screen with grey so we can see the effect of printing with and without
  // a background colour defined
 // tft.fillScreen(TFT_BLACK);

  // Set "cursor" at top left corner of display (0,0) and select font 2
  // (cursor will move to next line automatically during printing with 'tft.println'
  //  or stay on the line is there is room for the text with tft.print)
 // tft.setCursor(0, 0, 2);
  // Set the font colour to be white with a black background, set text size multiplier to 1
//  tft.fillScreen(TFT_BLACK); tft.setCursor(0, 0 ,2); tft.setTextColor(TFT_WHITE,TFT_BLACK);  tft.setTextSize(1); tft.println("Hello World!");
  // We can now plot text on screen using the "print" class
  tft.setTextColor(TFT_WHITE);  
  tft.setTextSize(2); 
  tft.setCursor(65, 50 ,2); 
  tft.print(Temperatur_IST); // Temperatur aus Variable auf das Display schreiben
  tft.setCursor(145, 50 ,2); 
  //tft.print("°C");  
  tft.print("`");        // Grad Celsius auf das Display schreiben
  tft.print("C");        // Grad Celsius auf das Display schreiben
  tft.setCursor(65, 80 ,2); 
  tft.print(Feuchte);        // Feuchte aus Variable auf das Display schreiben
  tft.setCursor(145, 80 ,2); 
  tft.print("%");           // % auf das Display schreiben

   //tft.pushImage(1, 60, PlusWidth, PlusHeight, Plus); // Icon auf Bildschirm darstellen
   //tft.pushImage(185, 60, MinusWidth, MinusHeight, Minus); // Icon auf Bildschirm darstellen
}

`

FYI you dont need to set the font size and colour each time, only when you need it to change.

My code looks like:

`          tft.setTextColor(TFT_WHITE,bgColour);  
          tft.setCursor(20, 50, 4);
          tft.setTextSize(1);
          tft.print("In: ");
          if(temp <= 19)
          {
            tft.setTextColor(TFT_BLUE,bgColour);
          }
          else if(temp > 19 && temp <= 23)
          {
            tft.setTextColor(TFT_GREEN,bgColour);
          }
          else
          {
            tft.setTextColor(TFT_RED,bgColour);
          }

          tft.print(temp);
          tft.setTextColor(TFT_WHITE,bgColour);  
          tft.print("`");
          tft.print("C");
`

quorle, 2019-04-25:
Ok, I tried your code, but unfortunately it still looks like this. (see picture) When I try your code, it makes me the degree Celsius wonderful. What can it be? Maybe all the other controls on the display?

[image]

elgerg, 2019-04-25:
Are you using   tft.setTextDatum in your code anywhere?

I suspect the degree symbol is too high up and too far to the right. Try removing your C letter and see if more of the degree symbol is printed.

With the numbers overwriting i think you need to need to clear the screen before writing to it.

Something like:
    tft.fillRect( 95, 80, 25, 20, TFT_BLACK );

quorle, 2019-04-25:
No, I do not use that tft.setTextDatum  in my code. 
Did put the symbol once in the middle of the display, but unfortunately does not change.

 What do you mean by the refresh of the display?

elgerg, 2019-04-25:
If you just write using tft.print(number) it wont clear the pixels from the number before.

Using tft.fillrect sets all of the pixels to black before you then write the new number.

Can you show me what it looks like without the "C" after?

Could mine be working because i am using font 4 with a size of 2?

Can you try this:

void display() {
// Fill screen with grey so we can see the effect of printing with and without
// a background colour defined
// tft.fillScreen(TFT_BLACK);

// Set "cursor" at top left corner of display (0,0) and select font 2
// (cursor will move to next line automatically during printing with 'tft.println'
// or stay on the line is there is room for the text with tft.print)
// tft.setCursor(0, 0, 2);
// Set the font colour to be white with a black background, set text size multiplier to 1
// tft.fillScreen(TFT_BLACK); tft.setCursor(0, 0 ,2); tft.setTextColor(TFT_WHITE,TFT_BLACK); tft.setTextSize(1); tft.println("Hello World!");
// We can now plot text on screen using the "print" class
tft.setTextColor(TFT_WHITE);
tft.setTextSize(1);
tft.setCursor(65, 50 ,4);
tft.print(Temperatur_IST); // Temperatur aus Variable auf das Display schreiben
tft.setCursor(145, 50 ,4);
//tft.print("°C");
tft.print("`"); // Grad Celsius auf das Display schreiben
tft.print("C"); // Grad Celsius auf das Display schreiben
tft.setCursor(65, 80 ,4);
tft.print(Feuchte); // Feuchte aus Variable auf das Display schreiben
tft.setCursor(145, 80 ,4);
tft.print("%"); // % auf das Display schreiben

//tft.pushImage(1, 60, PlusWidth, PlusHeight, Plus); // Icon auf Bildschirm darstellen
//tft.pushImage(185, 60, MinusWidth, MinusHeight, Minus); // Icon auf Bildschirm darstellen
}

quorle, 2019-04-25:
It really depends on the font size. With font size 2, the degree sign is no longer displayed. At font size 4 it works. Now only somehow the Refresh hinbekommen.

quorle, 2019-04-25:
Ok, now everything works, but can you explain the following to me and split what happens there and what you enter the numbers for?

`tft.fillRect( 95, 80, 25, 20, TFT_BLACK );`

quorle, 2019-04-25:
Ok, my code looks like this now. But it flickers at the point and the number after the integer at the temperature. Can one still switch off that somehow?

```
void display() {
// Fill screen with grey so we can see the effect of printing with and without
// a background colour defined
// tft.fillScreen(TFT_BLACK);

// Set "cursor" at top left corner of display (0,0) and select font 2
// (cursor will move to next line automatically during printing with 'tft.println'
// or stay on the line is there is room for the text with tft.print)
// tft.setCursor(0, 0, 2);
// Set the font colour to be white with a black background, set text size multiplier to 1
// tft.fillScreen(TFT_BLACK); tft.setCursor(0, 0 ,2); tft.setTextColor(TFT_WHITE,TFT_BLACK); tft.setTextSize(1); tft.println("Hello World!");
// We can now plot text on screen using the "print" class
/*tft.setTextColor(TFT_WHITE);
*/

tft.setCursor(60, 0, 4);
tft.print("Temperatur");
tft.setCursor(60, 50, 4);
tft.print("Ist:");
tft.setCursor(60, 80, 4);
tft.print("Soll:");
if (Temperatur_IST != 0)                     // Vergleich der aktuellen Temperatur auf ungleich Null, da nur Daten geschrieben werden sollen, wenn ein neuer Wert kommt
{
  tft.fillRect( 135, 50, 20, 20, TFT_BLACK ); // Aktuelle Bildpunkte bei neuem Wert auf Schwarz setzen, um kein Überschneiden des alten und neuen Wertes zu bekommen
                                              // Bedeutung der Zahlen: erste Zahl = Displayposition Breite / Zweite Zahl = Dislpayposition Höhe / Dritte Zahl = Länge / Vierte Zahl = Breite
  tft.setCursor(107, 50 ,4);                  // Cursor setzen für Anzeige aktuelle Temperatur
  tft.print(Temperatur_IST);                 // Temperatur aus Variable auf das Display schreiben
  tft.setCursor(160, 50 ,4);                 // Cursor setzen für °C Anzeige
  tft.print("`");                            // Grad Celsius auf das Display schreiben
  tft.print("C");                            // Grad Celsius auf das Display schreiben
}

if (Feuchte != 0)                            // Vergleich der aktuellen Feuchte auf ungleich Null, da nur Daten geschrieben werden sollen, wenn ein neuer Wert kommt
{
//  tft.fillRect( 125, 125, 26, 20, TFT_BLACK );  // Aktuelle Bildpunkte bei neuem Wert auf Schwarz setzen, um kein Überschneiden des alten und neuen Wertes zu bekommen
  tft.setCursor(125, 125 ,4);                   // Cursor setzen für Anzeige aktuelle Feuchte / Erste Zahl = Displayposition Breite / Zweite Zahl = Displayposition Höhe
  tft.print(Feuchte);                         // Feuchte aus Variable auf das Display schreiben
  tft.setCursor(160, 125 ,4);                  // Cursor setzen für % Anzeige
  tft.print("%");                             // % auf das Display schreiben
}
}
```

elgerg, 2019-04-26:
Ive seen this issue when updating the screen on every loop.

If i were you i would only change the screen when the values have changed.

So make a global variable, store the last value and compare that to the current.

Something like:

`float last_Temperatur_IST = 0.00;
float last_Feuchte = 0.00
void display() {
  // Fill screen with grey so we can see the effect of printing with and without
  // a background colour defined
  // tft.fillScreen(TFT_BLACK);
  
  // Set "cursor" at top left corner of display (0,0) and select font 2
  // (cursor will move to next line automatically during printing with 'tft.println'
  // or stay on the line is there is room for the text with tft.print)
  // tft.setCursor(0, 0, 2);
  // Set the font colour to be white with a black background, set text size multiplier to 1
  // tft.fillScreen(TFT_BLACK); tft.setCursor(0, 0 ,2); tft.setTextColor(TFT_WHITE,TFT_BLACK); tft.setTextSize(1); tft.println("Hello World!");
  // We can now plot text on screen using the "print" class
  /*tft.setTextColor(TFT_WHITE);
  */
  
  tft.setCursor(60, 0, 4);
  tft.print("Temperatur");
  tft.setCursor(60, 50, 4);
  tft.print("Ist:");
  tft.setCursor(60, 80, 4);
  tft.print("Soll:");
  if (Temperatur_IST != 0)                     // Vergleich der aktuellen Temperatur auf ungleich Null, da nur Daten geschrieben werden sollen, wenn ein neuer Wert kommt
  {
    if(last_Temperatur_IST != Temperatur_IST)
    {
      tft.fillRect( 135, 50, 20, 20, TFT_BLACK ); // Aktuelle Bildpunkte bei neuem Wert auf Schwarz setzen, um kein Überschneiden des alten und neuen Wertes zu bekommen
                                                  // Bedeutung der Zahlen: erste Zahl = Displayposition Breite / Zweite Zahl = Dislpayposition Höhe / Dritte Zahl = Länge / Vierte Zahl = Breite
      tft.setCursor(107, 50 ,4);                  // Cursor setzen für Anzeige aktuelle Temperatur
      tft.print(Temperatur_IST);                 // Temperatur aus Variable auf das Display schreiben
      tft.setCursor(160, 50 ,4);                 // Cursor setzen für °C Anzeige
      tft.print("`");                            // Grad Celsius auf das Display schreiben
      tft.print("C");                            // Grad Celsius auf das Display schreiben
      last_Temperatur_IST = Temperatur_IST;
    }
  }
  
  if (Feuchte != 0)                            // Vergleich der aktuellen Feuchte auf ungleich Null, da nur Daten geschrieben werden sollen, wenn ein neuer Wert kommt
  {
  //  tft.fillRect( 125, 125, 26, 20, TFT_BLACK );  // Aktuelle Bildpunkte bei neuem Wert auf Schwarz setzen, um kein Überschneiden des alten und neuen Wertes zu bekommen
    if(last_Feuchte != Feuchte)
    {
      tft.setCursor(125, 125 ,4);                   // Cursor setzen für Anzeige aktuelle Feuchte / Erste Zahl = Displayposition Breite / Zweite Zahl = Displayposition Höhe
      tft.print(Feuchte);                         // Feuchte aus Variable auf das Display schreiben
      tft.setCursor(160, 125 ,4);                  // Cursor setzen für % Anzeige
      tft.print("%");                             // % auf das Display schreiben
      last_Feuchte = Feuchte;
    {
  }
}`

Bodmer, 2019-05-01:
If you use drawString(), drawNumber() and drawFloat() you can use setPadding() to automatically overwite old digits and text with background. See TFT_Padding_demo example.

quorle, 2019-05-07:
Ok, thanks for your answer. I looked at it once, unfortunately I could not test it because the sketch is for the larger display. Therefore, I do not know yet, how I can integrate this into my project exactly.

Bodmer, 2019-05-07:
OK, assuming you have a 240x320 display look at the "Free_Font_Demo" example line 132 and "TFT_Custom_Fonts" line 128 onwards.

Bodmer, 2019-06-11:
Closed pending feedback.

closed 2019-06-11 by Bodmer

pikolo84, 2020-10-17:
Is late but if somebody need, print "÷" (dec 247)  solved the problem for me, i think ascii table is wrong. http://www.ltg.ed.ac.uk/~richard/utf-8.cgi?input=00B0&mode=hex

Rob4226, 2021-06-14:
Just want to add what took me a bit to figure out.  When **not** specifying a font size, what @pikolo84 said about using ÷ works  to get a ° degree symbol:
```c++
tft.print("÷");  // prints °
```
Although, I found that when specifying a font size number (like 2 in the following example), it does not work, it prints nothing. But then using a backtick \` works, like:
```c++
tft.drawString("`", 50, 100, 2);  // prints °
```

AmberElferink, 2024-02-10:
I wanted to work with tft.drawString instead of tft and the cursor. Additionally, none of the symbols above worked for me. My solution was to have a large list of every option I found on the internet so far to print to the screen. Then, I started changing the fonts:

  ```
#include <TFT_eSPI.h>
#include "Free_Fonts.h"

TFT_eSPI tft = TFT_eSPI();

  void setup()
  {
  tft.begin();
  tft.setRotation(1);
  tft.fillScreen(TFT_BLACK);
  tft.setTextColor(TFT_WHITE, tft.color565(5, 5, 5));
  tft.setFreeFont(FF18);

  tft.setTextSize(2); 
  tft.setTextPadding(tft.textWidth("8"));
  }
  
  
int fontNr = 0;
static uint32_t lastDrawTime = millis();
void loop() 
{

  float currentTemp = 1000;

  if (millis() - lastDrawTime >= 5000) {
    tft.setTextFont(fontNr);
    tft.setTextSize(1); 
    tft.drawNumber(fontNr, tft.getViewportWidth() - tft.textWidth("8"), 0);

    tft.drawString("÷C", 10, 0);
    tft.drawString("`C", 10, 20);
    tft.drawString(" \367""C", 10, 40);
    tft.drawString(" \337""C", 10, 60);
    tft.drawString(" \260""C", 10, 80);
    tft.drawString(" \xB0""C", 10, 100);

    fontNr++;

    //tft.drawString((String) (int) currentTemp + "/" + (String) (int) gs.getGoalTemp() + degreeSymbol, 10, 0);
    //tft.drawString(displayTimeRemaining(gs.getProgramDuration()), 10, 20);

    lastDrawTime = millis();
  }
  ```
  
  Just one look at this program and you can set a font you like. Within this loop somehow the fonts get larger each loop though, so you only have about 4 choices before they become too big You can probably try similar things for other fonts in free_fonts. But this simple script it solved my issue.

LEDFACIL, 2024-12-17:
> Just want to add what took me a bit to figure out. When **not** specifying a font size, what @pikolo84 said about using ÷ works to get a ° degree symbol:
> 
> ```c++
> tft.print("÷");  // prints °
> ```

Hey! that work f my, thanks!!!! funcionó perfecto, muchas gracias, un saludo desde Chile!