Bodmer/TFT_eSPI #351: Confirmation of touch screen connections

elgerg, 2019-04-24:
Hi,

Nice library works great on my 1.8 and 2.8 inc screens.

However.. I recently got an upgrade to my 2.8 inch screens, touch.

The driver chip is a "HR2046" XHR2046 clone?

Ive wired the 

T_CLK to the SCK pin
T_CS to D2
T_DO to D6

Unfortunately using the Test Touch Controller sketch i only get 0 for x and 0 for y... It the esp8266) doesnt seem to register any input. 

I've tested this with 2 screens so i assume i have either mis-wired or dont have enough set or the chip i have in not compatible..

Any help would be greatly appreciated.

Kind regards
Alex

elgerg, 2019-04-24:
The answer is to connect T_DIN to MOSI...

SO Connections as follows:

T_DO to D6 (MISO)
T_DIN to D7 (MOSI)
T_CS to D2
T_CLK to D5

Thanks

Bodmer, 2019-04-28:
Glad to here you have it working.
Thanks for the update.

closed 2019-04-28 by Bodmer