Bodmer/TFT_eSPI #352: Screen orientation setRotation(3) not inverting the x y co-ords

elgerg, 2019-04-24:
Hi,

Ive tried setting setting the screen rotation to 1 and 3, running and setting the calibration and cant seem to get the top left to be in the lower hundreds and bottom corner in the 4k range. These numbers would make sense if the screen was rotated to position 1 but not 3.

Am i missing some setting to invert the tft as well as the screen rotation?

Thanks
Alex

  tft.init();
  tft.setRotation(3);
  uint16_t calData[5] = { 200, 3359, 338, 3410, 7 };
  tft.setTouch(calData);

Bodmer, 2019-04-28:
The values do not need to be near 0 or 4000 and are dependant on the screen biasing.

Run the Test_touch_Controller diagnostic example to see the raw values at different screen positions, these values do not change when the drawing coordinates are rotated.

Noise can be a problem with these touch screens. 

The getTouch function already includes a noise filter, but some screens seem to be very noisy.

```
  uint16_t x = 0, y = 0; // To store the touch coordinates

  // Pressed will be set true is there is a valid touch on the screen
boolean pressed = tft.getTouch(&x, &y);
```

Bodmer, 2019-06-11:
Closed pending feedback.

closed 2019-06-11 by Bodmer