Bodmer/TFT_eSPI #354: Feature request: allow use of GPIO9/10 in OVERLAY mode

NdK73, 2019-04-28:
Hi.

On the ESP8266 it should be possible to use GPIO9 and GPIO10 to access display and touch devices **IFF** DIO (or DOUT) mode is used for internal flash. This way two more GPIOs remain available.
Else it would be correct to issue an #error at compile time.
I tried setting
```
#define ILI9341_DRIVER
#define TFT_SPI_OVERLAP
#warning "Ignoring comment about CS on pin D3"
#define TFT_CS 10
#define TFT_DC PIN_D4
#define TFT_RST  -1
```
and commenting out line 174 of TFT_eSPI.h:
```
// #undef TFT_CS
```
But still it does not work, so I'm surely missing something...
Any hint?

Tks.

NdK73, 2019-04-29:
PS: Also changed line 310 of TFT_eSPI.cpp to
```spi.pins(6, 7, 8, TFT_CS);```
But still no result :(

NdK73, 2019-04-29:
Found that spi.pins() call only accepts two configurations (6,7,8,0 for overlay, 14,12,13,* for HSPI).
Seems more changes (in the core, too) are needed.

Bodmer, 2019-05-01:
If you find a solution then let me know but this is not something I plan to investigate or test and if changes to the core are needed I am unable to help.

closed 2019-05-01 by Bodmer