Bodmer/TFT_eSPI #356: Extend font 7

Petros144, 2019-05-05:
Hi!

The Font 7 that comes with this library fits exactly my needs, but I would add 2 - 3 extra Characters. Its not GFX encoded so i cant load it into the editior.

If someone knows the font or a way to modify the Font file, please tell me.

files are:

Font7srle.c
Font7srle.h

Bodmer, 2019-05-07:
The font file is run length encoded so it is very compact, this was intended for memory constrained processors like the Arduino.  Unfortunately the price paid for this is that it makes adding characters a complex and time consuming process.

Petros144, 2019-05-07:
Thanks for the answer Bodmer! 
I think i will make my own font, will be easier.

Before I start a second issue, you can maybe answer this:

If I run the ili9341 with a short extension cable (30cm / 1ft), everytime the ESP32 wirites to the display the backlight gets a little bit brighter. I tought that it is  solved with a capacitor on VCC /LED to ground. but this was not the case?! 

Is it the CS line that make this effect?

Bodmer, 2019-05-08:
It is likely your display has a 3.3V regulator fitted, powering this from 3.3V would mean the regulator maybe only outputs 3V. The logic levels will be near 3.3V so maybe they are lifting the VCC rail slightly causing more backlight current.

If your display does indeed have a 3.3V regulator then you can power the VCC line from 5V.

closed 2019-05-08 by Bodmer