Bodmer/TFT_eSPI #357: Subscript characters not displaying

thearduinoguy, 2019-05-07:
I have enabled the superscript range of characters in the Create_font.pde file but they are still displaying as rectangles. Please help.

Bodmer, 2019-05-07:
If they are displzayed on your PC as rectangles then it is because the font you are using does not contain the subscript characters. Check the truetype font includes the character set by checking the web or font documentation.

thearduinoguy, 2019-05-07:
Hi, I am using Noto Sans Extra Light. Using Character Map I can see that this font does have the subscript numbers. I am trying to use 0x2082 (subscript 2) and 0x2084 (subscript 4). Both are appearing as rectangles.

thearduinoguy, 2019-05-07:
I have created the characters in the code using 👍 
`#define micro_symbol String(char(0x00B5))
#define cubed_symbol String(char(0x00B3))
#define subscript_two String(char(0x2082))
#define subscript_four String(char(0x2084))`

The micro symbol and cubed symbol work fine, the two subscript characters are both rectangles

In the Create_font.pde program I have uncommented the line:

`  0x2070, 0x209F, //Superscripts and Subscripts, 48, 42, Latin (15 characters), Common (27 characters)`

Bodmer, 2019-05-07:
Unicode characters are 16 bits so have to be defined like this:

#define subscript_two "\u2082"

or:

tft.print("\u2084");

The \u instructs the compiler to UTF-8 encode the value, this then gets encoded as 3 bytes.

thearduinoguy, 2019-05-07:
OK the `#define` did not work, it threw an error. But when I did `"CO\u2082"` for CO₂ it worked. Thanks.

Bodmer, 2019-05-07:
The may erroneously point at the #define when in fact the error is where it is used. If you simply try to add the strings you will get an error, this is the way to handle that:

```
  #define subscript_two "\u2082"

  tft.println( "CO" + (String)subscript_two );
```

closed 2019-05-08 by Bodmer