Bodmer/TFT_eSPI #358: Show image on TFT screen from REST Request

OmarBochkati, 2019-05-08:
Hello,

Thank you for this awesome library it helped a lot in my project.
But i'm stuck. What I am trying to accomplish is : 
 * Send image to esp32
 * Convert the image to the right type/format
 * Display the image on screen

I tried converting the image before hand using the website you mentioned in comments [http://tomeko.net/online_tools/file_to_hex.php?lang=en](http://tomeko.net/online_tools/file_to_hex.php?lang=en) and I used "ESPAsyncWebServer" to create small rest server on the ESP32 and i recieved the image data correctly but i struggle in displaying it.

> server.on("/api/updateImage", HTTP_POST, [](AsyncWebServerRequest *request){
                   if (request->hasParam("image", true)) {
                   Serial.println(request->getParam("image", true)->value().c_str());
                  tft.drawBitmap(0,0,(const uint8_t*)request->getParam("image", true)->value().c_str(), 100,100, TFT_BLACK);
                  delay(1000);
                 tft.drawXBitmap(0, 0, (const uint8_t*)request->getParam("image", true)->value().c_str(), 100, 100, TFT_WHITE);
                delay(1000);
                //drawArrayJpeg((const uint8_t*)request->getParam("image", true)->value().c_str(), 
               sizeof((const uint8_t*)request->getParam("image", true)->value().c_str()), 1, 1);
              request->send(200, "text/plain", "Image updated");
              } 
              request->send(400, "text/plain", "Image not updated");
      });

I can see on the serial monitor that the data arrieved, but non of the methods i used display the data correctly.
Please help me, or point me to where I messed up.

Thank you so much

Bodmer, 2019-05-08:
drawBitmap and drawXBitmap are for 2 colour images where the image pixels are encoded as bytes.  The drawArrayJpeg function expects an array of bytes. It is not clear what format you are converting the image into.

I suggest you use image data from the examples and then the sketch could compare what it receives with the original file then you will be able to see where things are going wrong.

Unfortunately I am unable to advise further as this is not liekly to be an issue with the TFT_eSPI library.

OmarBochkati, 2019-05-09:
Thank you for your answer !

I tried examples you provided, images data that I also converted and tried for both drawBitmap and drawArrayJpeg. The converted data work if it is loaded in a header file but don't work when I send them over an HTTP POST request.

Bodmer, 2019-05-19:
The data received from the HTTP POST must be in a different format. This is not something I can help you with.

closed 2019-05-19 by Bodmer

jishan484, 2020-04-16:
i have a working project like you .
Server site :
i use image of 128 x 240 and convert it using xxd -i -h command
then using nodejs i send that data to esp32
 
code esp32:
#include <TFT_eSPI.h>
#include <WiFi.h>
#include <stdlib.h>
#include <JPEGDecoder.h>
#include <pgmspace.h>
#include <Button2.h>

#ifndef TFT_DISPOFF
#define TFT_DISPOFF 0x28
#endif

#ifndef TFT_SLPIN
#define TFT_SLPIN   0x10
#endif

#define minimum(a,b)     (((a) < (b)) ? (a) : (b))
#define TFT_MOSI            19
#define TFT_SCLK            18
#define TFT_CS              5
#define TFT_DC              16
#define TFT_RST             23

#define BUTTON_1        35
#define BUTTON_2        0
#define TFT_BL          4

TFT_eSPI tft = TFT_eSPI(135, 240); // Invoke custom library
Button2 btn1(BUTTON_1);
Button2 btn2(BUTTON_2);
int btnCick = false;
const char* ssid     = "jishan";
const char* password = "jishan007";
const char* host = "outrageous-crustacean.glitch.me";
uint8_t image[20480] PROGMEM={0};
void setup()
{
    tft.init();
    tft.setRotation(0);
    tft.fillScreen(TFT_BLACK);
    setwifi();
    button_init();
}

void setwifi()
{
    WiFi.begin(ssid, password);
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
    }
    delay(5000);
}

void loop()
{
        show_image();
        button_loop();
        delay(10000);
}
void show_image()
{
  WiFiClient client;
    const int httpPort = 80;
    if (!client.connect(host, httpPort)) {
      delay(4000);
      show_image();
    }
    String url = "/";
    client.print(String("GET ") + url + " HTTP/1.1\r\n" +
                 "Host: " + host + "\r\n" +
                 "Connection: close\r\n\r\n");
    unsigned long timeout = millis();
    while (client.available() == 0) {
        if (millis() - timeout > 5000) {
            client.stop();
        }
    }
    String line;
    int a=0;
    while(client.available()) {
        line = client.readStringUntil('\r');
    }
     char delimiter[] = ",";
    char* ptr = strtok(&line[1], delimiter);
    while(ptr != NULL) {
        image[a] = (int) strtol( ptr, NULL, 16);
        a++;
        ptr = strtok(NULL, delimiter);
    }
    drawArrayJpeg(image, sizeof(image), 0, 0);
}

void button_init()
{
    btn1.setLongClickHandler([](Button2 & b) {
        btnCick = false;
        int r = digitalRead(TFT_BL);
        tft.fillScreen(TFT_BLACK);
        tft.setTextColor(TFT_GREEN, TFT_BLACK);
        tft.setTextDatum(MC_DATUM);
        tft.drawString("Press again to wake up",  tft.width() / 2, tft.height() / 2 );
        delay(6000);
        digitalWrite(TFT_BL, !r);

        tft.writecommand(TFT_DISPOFF);
        tft.writecommand(TFT_SLPIN);
        esp_sleep_enable_ext1_wakeup(GPIO_SEL_35, ESP_EXT1_WAKEUP_ALL_LOW);
        esp_deep_sleep_start();
    });
    btn1.setPressedHandler([](Button2 & b) {
        esp_sleep_enable_ext1_wakeup(GPIO_SEL_35, ESP_EXT1_WAKEUP_ALL_LOW);
        esp_deep_sleep_start();
    });

    btn2.setPressedHandler([](Button2 & b) {
        esp_sleep_enable_ext1_wakeup(GPIO_SEL_35, ESP_EXT1_WAKEUP_ALL_LOW);
        esp_deep_sleep_start();
    });
}
void button_loop()
{
    btn1.loop();
    btn2.loop();
}

void drawArrayJpeg(const uint8_t arrayname[], uint32_t array_size, int xpos, int ypos) {
  int x = xpos;
  int y = ypos;
  JpegDec.decodeArray(arrayname, array_size);
  renderJPEG(x, y);
}
void renderJPEG(int xpos, int ypos) {
  uint16_t *pImg;
  uint16_t mcu_w = JpegDec.MCUWidth;
  uint16_t mcu_h = JpegDec.MCUHeight;
  uint32_t max_x = JpegDec.width;
  uint32_t max_y = JpegDec.height;
  uint32_t min_w = minimum(mcu_w, max_x % mcu_w);
  uint32_t min_h = minimum(mcu_h, max_y % mcu_h);
  uint32_t win_w = mcu_w;
  uint32_t win_h = mcu_h;
  max_x += xpos;
  max_y += ypos;
  while (JpegDec.readSwappedBytes()) {
    pImg = JpegDec.pImage ;
    int mcu_x = JpegDec.MCUx * mcu_w + xpos;  // Calculate coordinates of top left corner of current MCU
    int mcu_y = JpegDec.MCUy * mcu_h + ypos;
    if (mcu_x + mcu_w <= max_x) win_w = mcu_w;
    else win_w = min_w;
    if (mcu_y + mcu_h <= max_y) win_h = mcu_h;
    else win_h = min_h;
    if (win_w != mcu_w)
    {
      uint16_t *cImg;
      int p = 0;
      cImg = pImg + win_w;
      for (int h = 1; h < win_h; h++)
      {
        p += mcu_w;
        for (int w = 0; w < win_w; w++)
        {
          *cImg = *(pImg + w + p);
          cImg++;
        }
      }
    }
    if (( mcu_x + win_w ) <= tft.width() && ( mcu_y + win_h ) <= tft.height())
    {
      tft.pushRect(mcu_x, mcu_y, win_w, win_h, pImg);
    }
    else if ( (mcu_y + win_h) >= tft.height()) JpegDec.abort();
  }
}