Bodmer/TFT_eSPI #359: Some artifacts redering JPEG in a sprite

JavierPascual, 2019-05-10:
Hi @Bodmer!!

I've doing more tests and with some little JPEG files, I'm seeing new artifacts in screen on the right side of the images. I'm loading and painting it with:

`lcd_fx.drawJpeg("/heart.jpg", x, v, _buffer_sprite);`

Some real examples here, in a ILI9341 320x240 display:

[image]
[image]
[image]

_Originally posted by @JavierPascual in https://github.com/Bodmer/TFT_eSPI/issues/289#issuecomment-490816157_

Bodmer, 2019-05-10:
Thanks for reporting this.

I see an issue in the function, due to some urgent work I will not be able to fix it and test for 2 weeks.

JavierPascual, 2019-05-11:
Great! thank you Bodmer! 👍 👍 👍

Bodmer, 2019-05-20:
This was a bug in the TFT_eFEX library that has now been corrected. Please download and try the latest version.

JavierPascual, 2019-05-22:
Hi @Bodmer! Yes, the bug has been fixed! Thank you very much!!

closed 2019-05-22 by Bodmer