Bodmer/TFT_eSPI #360: Strange logic drawIcon() function on ST7789 240x240 TFT

fa1ke5, 2019-05-10:
i'm using drawIcon() function on ST7789 240x240 TFT but i see strange effect, 
i have correct logo picture:
// Generated by  : ImageConverter 565 v2.2
// Generated from: mark_logo_big.bmp
// Time generated: 23.01.2019 23:07:20
// Dimensions    : 198x235 pixels
// Size          : 93 060 Bytes

const unsigned short mark_logo_big[0xB5C2] ={
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000 e.t.c

and use fuction as is
drawIcon(mark_logo_big,  1, 1, 198,  235);
it work nice, but i im change X or Y position picture is corrupting.
for example 
drawIcon(mark_logo_big,  10, 1, 198,  235);
give me picture like broken CRT TV.
if im changin dimensions of picture as short as X long, picture is correct now
drawIcon(mark_logo_big,  10-1, 1, 198-10,  235+10);

Bodmer, 2019-05-10:
Which example sketch are you using?

fa1ke5, 2019-05-10:
oh, thanks, i found problem drawIcon() is non library function!
closed...

closed 2019-05-10 by fa1ke5