Bodmer/TFT_eSPI #361: E_spi V1.4.8  Compilation error for Weemos D1 mini board or jnodemcu

matthieuweber, 2019-05-17:
Hello, I would try your library but I got this error when compiling it :+1:  
Seems it looks like this issue: https://github.com/Bodmer/TFT_eSPI/issues/15)

Thanks for your helps !

Arduino : 1.8.9 (Windows 7), Carte : "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (3M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Les options de compilation ont été modifiées, tout sera recompilé
In file included from C:\Users\mweber-admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1\cores\esp8266/pgmspace.h:7:0,

                 from C:\Users\mweber-admin\Documents\Arduino\libraries\TFT_eSPI/Fonts/glcdfont.c:6,

                 from C:\Users\mweber-admin\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.h:62,

                 from C:\Users\mweber-admin\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:17:

C:\Users\mweber-admin\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'virtual int16_t TFT_eSPI::drawChar(uint16_t, int32_t, int32_t, uint8_t)':

C:\Users\mweber-admin\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:4262:84: error: invalid conversion from 'uint32_t {aka unsigned int}' to 'const void*' [-fpermissive]

       flash_address = pgm_read_dword( pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *) );

                                                                                    ^

C:\Users\mweber-admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:114:57: note: in definition of macro 'pgm_read_dword'

     #define pgm_read_dword(a)  pgm_read_dword_unaligned(a)

                                                         ^

C:\Users\mweber-admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:85:72: error:   initializing argument 1 of 'uint32_t pgm_read_dword_unaligned(const void*)' [-fpermissive]

 __attribute__((optimize("-O3"), always_inline)) static inline uint32_t pgm_read_dword_unaligned(const void *addr) {

                                                                        ^

C:\Users\mweber-admin\Documents\Arduino\libraries\TFT_eSPI\Extensions/Sprite.cpp: In member function 'virtual int16_t TFT_eSprite::drawChar(uint16_t, int32_t, int32_t, uint8_t)':

C:\Users\mweber-admin\Documents\Arduino\libraries\TFT_eSPI\Extensions/Sprite.cpp:1682:84: error: invalid conversion from 'uint32_t {aka unsigned int}' to 'const void*' [-fpermissive]

       flash_address = pgm_read_dword( pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *) );

                                                                                    ^

C:\Users\mweber-admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:114:57: note: in definition of macro 'pgm_read_dword'

     #define pgm_read_dword(a)  pgm_read_dword_unaligned(a)

                                                         ^

C:\Users\mweber-admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:85:72: error:   initializing argument 1 of 'uint32_t pgm_read_dword_unaligned(const void*)' [-fpermissive]

 __attribute__((optimize("-O3"), always_inline)) static inline uint32_t pgm_read_dword_unaligned(const void *addr) {
                                                                      ^

exit status 1

d-a-v, 2019-05-17:
```
   flash_address = pgm_read_dword( (const void*)pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *) );
```

luc-github, 2019-05-17:
Looks same as https://github.com/Bodmer/TFT_eSPI/issues/348

d-a-v, 2019-05-17:
About #348, esp8266/arduino 2.5.1 is officially out now.
If there is still an issue on our side, @luc-github @Bodmer please tell us, we are planning 2.5.2 for next week with some important updates following 2.5.1.

jpasqua, 2019-05-17:
> Try with this:
> 
> ```
>    flash_address = pgm_read_dword( (const void*)pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *) );
> ```

That fixed the compile issue for me.

Bodmer, 2019-05-19:
Thanks, I have added the update line.

closed 2019-05-19 by Bodmer

matthieuweber, 2019-05-20:
Thanks a lots, I replaced in Sprite.cpp and TFT_eSPI.cpp files line   
flash_address = pgm_read_dword( pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *) );
with 

 flash_address = pgm_read_dword( (const void*)pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *) );

Compilation is now OK :)

By the way, It worked without this modification on another computer (same Arduino IDE & lib version but win 10...) Funny !

Very impressive library !  Great job !  :o)

Matthieu

Bodmer, 2019-05-20:
Thanks, have updated Sprite.cpp, I forgot that one!