Bodmer/TFT_eSPI #362: Wiring and White screen MCUFriend 3.5 TFT connect NodeMCU ESP8266

muriox, 2019-05-18:
Hi Bodmer,

Please I'm trying to MCUFriend 3.5 TFT for arduino UNO to NodeMCU ESP8266 and also use the SD card for M.Sc Project. 

See [MCUFriend 3.5 TFT](https://www.picclickimg.com/00/s/ODAwWDgwMA==/z/i4sAAOSwB-1YnDT5/$/35-inch-TFT-LCD-Display-Arduino-Touch-Screen-_57.jpg)

Following the guide in User_Setup.h:

_// We must use hardware SPI, a minimum of 3 GPIO pins is needed.
// Typical setup for ESP8266 NodeMCU ESP-12 is :
//
// Display SDO/MISO  to NodeMCU pin D6 (or leave disconnected if not reading TFT)
// Display LED       to NodeMCU pin VIN (or 5V, see below)
// Display SCK       to NodeMCU pin D5
// Display SDI/MOSI  to NodeMCU pin D7
// Display DC (RS/AO)to NodeMCU pin D3
// Display RESET     to NodeMCU pin D4 (or RST, see below)
// Display CS        to NodeMCU pin D8 (or GND, see below)
// Display GND       to NodeMCU pin GND (0V)
// Display VCC       to NodeMCU 5V or 3.3V_

I'm still getting the white screen.
No **Display LED** pin on my MCUFriend display.

Where should connect:
LCD_RD
LCD_WR
and 
LCD_D0 - LCD_D7 ?

Your guide will be highly appreciated

Bodmer, 2019-05-18:
You have an 8bit parallel screen. The library only supports ESP32 boards in parallel mode.

muriox, 2019-05-18:
Sorry for disturbing you. 
I'm kind of new to this. Please is there any way I could make it work or direction you can point me?

Regards

Bodmer, 2019-05-18:
One option is to buy an UNO format ESP32 board and modify as in ReadMe.

Another option is to buy a compatible SPI based TFT screen.

closed 2019-05-19 by muriox