Bodmer/TFT_eSPI #364: What would be the best way to add dual screen support?

yam001, 2019-05-19:


yam001, 2019-05-19:
Thanks very much for providing such a very nice library and it works very well. It's quite well done. Using this library, I have managed to pull images off of camera servers and display them on ESP32s. I am currently using 6 of these, one for each camera. I have looked at your code a little bit and thinking about adding some kind of hooks to allow graphics operations to be localized to one SPI channel so that two screens can work at the same time, one for VSPI, and one for HSPI concurrently. Do you have any thoughts on how this could be accomplished?

yam001, 2019-05-19:
The part that makes it not easily modifiable is the many uses of #defines.

Bodmer, 2019-05-19:
As you note the library does not support simultaneous use of the VSPI and HSPI ports.

One option is to connect the two displays to one SPI port, use one instance of the library and select the display to write to by controlling each chip select in the sketch.

closed 2019-06-11 by Bodmer