Bodmer/TFT_eSPI #365: Incorrect touch remap following setRotation()

ImpulseAdventure, 2019-05-21:
Thanks again for the great work on TFT_eSPI!

In testing the TFT_eSPI **integrated XPT2046 touch handler**, I noticed that the touch remapping (per calibration) does not appear to be operating as expected following rotation.

The default calibration (**Touch_calibrate**) captures the parameter set from rotation=1. The Touch_calibrate sketch demonstrates correct touch mapping to the display in this same orientation (rotation=1).

### Observation
However, if `getTouch()` is called after updating the display orientation via `setRotation()`, the touch remapping no longer appears to be rotated correctly:
- **rotation=0**: Touch is rotated -90deg clockwise
- **rotation=1**: Matched
- **rotation=2**: Touch is rotated +90deg clockwise
- **rotation=3**: Touch is rotated +180deg clockwise

As you are aware, other displays may exhibit a different remapping transformation than reflected above.

### Testcase
I have attached a modified **Touch_calibrate-rotate** sketch to help you recreate the issue. After 200 points are collected, the display orientation is advanced.
[Touch_calibrate-rotate.zip](https://github.com/Bodmer/TFT_eSPI/files/3202863/Touch_calibrate-rotate.zip)

### Background details
- **MCU**: SparkFun ESP32 Thing
- **TFT**: 2.8" ILI9341 breakout board
- **Library version**: TFT_eSPI 1.4.10 (da2239f)
- **Calibration result**: `uint16_t calData[5] = { 485, 3382, 316, 3419, 3 };`

Thanks

Bodmer, 2019-06-03:
The intent was that the screen should be calibrated for the rotation you intend to use. The library does not handle dynamic rotation of the touch coordinates.

I do not plan to update the library at the moment.

closed 2019-06-03 by Bodmer

ImpulseAdventure, 2019-06-07:
Thanks Bodmer for confirming the operation. I had assumed that the functionality would be left as-is, given that the hardware variability in touch remapping to display rotation can make for a non-trivial implementation!