Bodmer/TFT_eSPI #367: BRG to RGB in function pushImage with ili9486, ili9488

ali-batkhed, 2019-05-26:
If you have shifted color palette from RGB to BRG with ili9486 driver in parallel 8-bit mode - you should remove swapping bytes in this function.

in line 1010 was:

  while (dh--)
  {
    pushColors(data, dw, _swapBytes);
    data += w;
  }

, must be:

  while (dh--)
  {
    pushColors(data, dw);
    data += w;
  }.

Good luck to everyone!

Bodmer, 2019-05-26:
Do not modify the code, use the setSwapBytes(true/false) function in your sketch.

by default _swapBytes is set to false. See [prototypes here](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.h#L762-L764).

closed 2019-05-26 by Bodmer

ali-batkhed, 2019-06-04:
thank you very much! i'm a beginner in arduino - that's why i missed.