Bodmer/TFT_eSPI #368: compilation errors in esp8266 arduino core 2.5.2

jmcastillejo, 2019-05-26:
Hi! in de last core version the compilation fails:

In file included from /home/jose/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/pgmspace.h:7:0,
                 from /home/jose/Arduino/libraries/TFT_eSPI-master/Fonts/glcdfont.c:6,
                 from /home/jose/Arduino/libraries/TFT_eSPI-master/TFT_eSPI.h:62,
                 from /home/jose/Arduino/libraries/TFT_eSPI-master/TFT_eSPI.cpp:17:
/home/jose/Arduino/libraries/TFT_eSPI-master/TFT_eSPI.cpp: In member function 'virtual int16_t TFT_eSPI::drawChar(uint16_t, int32_t, int32_t, uint8_t)':
/home/jose/Arduino/libraries/TFT_eSPI-master/TFT_eSPI.cpp:4274:120: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
       flash_address = pgm_read_dword( (const void*)pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *) );
                                                                                                                        ^
/home/jose/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:114:57: note: in definition of macro 'pgm_read_dword'
     #define pgm_read_dword(a)  pgm_read_dword_unaligned(a)
                                                         ^
/home/jose/Arduino/libraries/TFT_eSPI-master/Extensions/Sprite.cpp: In member function 'virtual int16_t TFT_eSprite::drawChar(uint16_t, int32_t, int32_t, uint8_t)':
/home/jose/Arduino/libraries/TFT_eSPI-master/Extensions/Sprite.cpp:1682:84: error: invalid conversion from 'uint32_t {aka unsigned int}' to 'const void*' [-fpermissive]
       flash_address = pgm_read_dword( pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *) );
                                                                                    ^
/home/jose/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:114:57: note: in definition of macro 'pgm_read_dword'
     #define pgm_read_dword(a)  pgm_read_dword_unaligned(a)
                                                         ^
/home/jose/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:85:72: error:   initializing argument 1 of 'uint32_t pgm_read_dword_unaligned(const void*)' [-fpermissive]
 __attribute__((optimize("-O3"), always_inline)) static inline uint32_t pgm_read_dword_unaligned(const void *addr) {
                                                                        ^
exit status 1
Error compilando para la tarjeta NodeMCU 1.0 (ESP-12E Module).

With version 2.5.0 compile without errors

Bodmer, 2019-05-26:
This has been fixed, see #361. New version of TFT_eSPI is 1.4.10 and will be avialable to the Arduino IDE library manager within 24hrs.

jmcastillejo, 2019-05-26:
I have updated the library, compile but the warnings continue:

In file included from /home/jose/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/pgmspace.h:7:0,
                 from /home/jose/Arduino/libraries/TFT_eSPI-master/Fonts/glcdfont.c:6,
                 from /home/jose/Arduino/libraries/TFT_eSPI-master/TFT_eSPI.h:62,
                 from /home/jose/Arduino/libraries/TFT_eSPI-master/TFT_eSPI.cpp:17:
/home/jose/Arduino/libraries/TFT_eSPI-master/TFT_eSPI.cpp: In member function 'virtual int16_t TFT_eSPI::drawChar(uint16_t, int32_t, int32_t, uint8_t)':
/home/jose/Arduino/libraries/TFT_eSPI-master/TFT_eSPI.cpp:4274:120: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
       flash_address = pgm_read_dword( (const void*)pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *) );
                                                                                                                        ^
/home/jose/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:114:57: note: in definition of macro 'pgm_read_dword'
     #define pgm_read_dword(a)  pgm_read_dword_unaligned(a)
                                                         ^
/home/jose/Arduino/libraries/TFT_eSPI-master/Extensions/Sprite.cpp: In member function 'virtual int16_t TFT_eSprite::drawChar(uint16_t, int32_t, int32_t, uint8_t)':
/home/jose/Arduino/libraries/TFT_eSPI-master/Extensions/Sprite.cpp:1683:120: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
       flash_address = pgm_read_dword( (const void*)pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *) );
                                                                                                                        ^
/home/jose/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:114:57: note: in definition of macro 'pgm_read_dword'
     #define pgm_read_dword(a)  pgm_read_dword_unaligned(a)

KuhnDaniel, 2019-06-02:
Hello Herr Bodmer
I have updated the library, compile but the warnings continue: 1.4.10 2.6.2019
TFT_eSPI/Extensions/Sprite.cpp:1683:120: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
       flash_address = pgm_read_dword( (const void*)pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *) );

Bodmer, 2019-06-03:
I have created a new release 1.4.11 that corrects this. It will be available from the IDE library manager within 24 hours.

Thanks for reporting it.

closed 2019-06-03 by Bodmer

KuhnDaniel, 2019-06-04:
Thanks for your good Job

KuhnDaniel, 2019-06-04:
good morning, the result of last compilation: V1.4.11 /ESP32 1.0.3-rc1

In file included from /home/daniel/sketchbook/libraries/TFT_eSPI/Fonts/glcdfont.c:6:0,
                 from /home/daniel/sketchbook/libraries/TFT_eSPI/TFT_eSPI.h:62,
                 from /home/daniel/sketchbook/libraries/TFT_eSPI/TFT_eSPI.cpp:17:
/home/daniel/sketchbook/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'virtual int16_t TFT_eSPI::drawChar(uint16_t, int32_t, int32_t, uint8_t)':
/home/daniel/sketchbook/libraries/TFT_eSPI/TFT_eSPI.cpp:4273:120: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
       flash_address = pgm_read_dword( (const void*)pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *) );
                                  ^

Bodmer, 2019-06-04:
OK, the original issue has been solved but this appears to have impacted the ESP32. Can you raise this as a new issue for specifically for the ESP32? In the meantime I will look at the cause.
Thanks.