Bodmer/TFT_eSPI #369: Sprite.cpp: error: invalid conversion from 'uint32_t {aka unsigned int}' to 'const void*'

a-x-, 2019-05-26:
$HOME/Documents/Arduino/libraries/TFT_eSPI/Extensions/Sprite.cpp:1682:84: error: invalid conversion from 'uint32_t {aka unsigned int}' to 'const void*' [-fpermissive]

TFT_eSPI lib version: latest (1.4.8) — didn't use before
wemos d1r2
platformio 3.6.7

a-x-, 2019-05-26:
<details><summary>log</summary>

```
Processing d1_mini (platform: espressif8266; board: d1_mini; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Converting esp-ST7735-display.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 46 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SPI> 1.0
|-- <TFT_eSPI> 1.4.8
|   |-- <SPI> 1.0
Compiling .pioenvs/d1_mini/src/esp-ST7735-display.ino.cpp.o
Compiling .pioenvs/d1_mini/libbd0/TFT_eSPI/TFT_eSPI.cpp.o
In file included from /Users/mxtnr/Documents/PlatformIO/Projects/190526-180643-d1_mini/src/esp-ST7735-display.ino:12:0:
/Users/mxtnr/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.h:665:25: error: 'TFT_WIDTH' was not declared in this scope
TFT_eSPI(int16_t _W = TFT_WIDTH, int16_t _H = TFT_HEIGHT);
^
/Users/mxtnr/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.h:665:49: error: 'TFT_HEIGHT' was not declared in this scope
TFT_eSPI(int16_t _W = TFT_WIDTH, int16_t _H = TFT_HEIGHT);
^
/Users/mxtnr/Documents/PlatformIO/Projects/190526-180643-d1_mini/src/esp-ST7735-display.ino:16:25: error: call to 'TFT_eSPI::TFT_eSPI(int16_t, int16_t)' uses the default argument for parameter 1, which is not yet defined
TFT_eSPI tft = TFT_eSPI();  // Invoke library, pins defined in User_Setup.h
^
/Users/mxtnr/Documents/PlatformIO/Projects/190526-180643-d1_mini/src/esp-ST7735-display.ino:16:25: error: call to 'TFT_eSPI::TFT_eSPI(int16_t, int16_t)' uses the default argument for parameter 2, which is not yet defined
*** [.pioenvs/d1_mini/src/esp-ST7735-display.ino.cpp.o] Error 1
/Users/mxtnr/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'void TFT_eSPI::setWindow(int32_t, int32_t, int32_t, int32_t)':
/Users/mxtnr/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:2695:22: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]
SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);
^
/Users/mxtnr/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:2695:47: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]
SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);
^
/Users/mxtnr/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:2716:22: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]
SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);
^
/Users/mxtnr/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:2716:47: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]
SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);
^
/Users/mxtnr/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:2733:22: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]
SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);
^
/Users/mxtnr/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:2733:47: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]
SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);
^
/Users/mxtnr/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'void TFT_eSPI::readAddrWindow(int32_t, int32_t, int32_t, int32_t)':
/Users/mxtnr/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:2953:22: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]
SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);
^
/Users/mxtnr/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:2953:47: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]
SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);
^
/Users/mxtnr/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:2970:22: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]
SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);
^
/Users/mxtnr/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:2970:47: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]
SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);
^
/Users/mxtnr/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:2987:22: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]
SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);
^
/Users/mxtnr/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:2987:47: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]
SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);
^
[... 33 lines cut ...]
In file included from /Users/mxtnr/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:5327:0:
/Users/mxtnr/Documents/Arduino/libraries/TFT_eSPI/Extensions/Sprite.cpp: In member function 'void TFT_eSprite::pushImage(int32_t, int32_t, int32_t, int32_t, const uint16_t*)':
/Users/mxtnr/Documents/Arduino/libraries/TFT_eSPI/Extensions/Sprite.cpp:719:31: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
for (int32_t xp = 0; xp<ww; xp+=8)
^
In file included from /Users/mxtnr/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/pgmspace.h:7:0,
from /Users/mxtnr/Documents/Arduino/libraries/TFT_eSPI/Fonts/glcdfont.c:6,
from /Users/mxtnr/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.h:62,
from /Users/mxtnr/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:17:
[... cut ...]
/Users/mxtnr/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:85:72: error:   initializing argument 1 of 'uint32_t pgm_read_dword_unaligned(const void*)' [-fpermissive]
__attribute__((optimize("-O3"), always_inline)) static inline uint32_t pgm_read_dword_unaligned(const void *addr) {
^
*** [.pioenvs/d1_mini/libbd0/TFT_eSPI/TFT_eSPI.cpp.o] Error 1
=============================================================== [ERROR] Took 1.85 seconds ===============================================================
The terminal process terminated with exit code: 1
```

</details>

a-x-, 2019-05-26:
related  issue: https://github.com/Bodmer/TFT_eSPI/issues/368

seems the same.

Bodmer, 2019-05-26:
This has been fixed, see #361. New version of TFT_eSPI is 1.4.10 and will be avialable to the Arduino IDE library manager within 24hrs.

closed 2019-05-26 by a-x-