Bodmer/TFT_eSPI #370: ESP32 with SPI ILI9488 Clock and Text Scrolling

Dancopy, 2019-05-27:
I would like to suggest to Bodmer, through its availability, to adapt its 2 "NTP_Analog_clk_tz" and "NTP_Digital_clk_tz" sketches from its old library to its current TFT_eSPI library for a 3.5" ILI9488 display with ESP32 or ESP8266, even leaving Also, add as an example, Text with Vertical and Horizontal Scrolling. Thanks to Bodmer

Bodmer, 2019-06-02:
I am not aware of any NTP_Analog_clk_tz examples. The Sprite examples demo scrolling in small windows.

closed 2019-06-02 by Dancopy

Dancopy, 2019-08-31:
I was wrong! This is the TFT_Clock sketch of the 320 x 240 folder, but I would like to adapt it to the 3.5" Display SPI Breakout ILI9488 480 x 320 with ESP32 or ESP8266 Nodemcu.
I love these Watches, especially Analog!
I would be very grateful for any help!