Bodmer/TFT_eSPI #371: Progmem image is drawing at a 45deg angle...

clivemist, 2019-05-29:
Hi. I am using arduino.

Today it auto-updated. Everything ran fine earlier today, but now "drawicon" displays images at a 45deg angle from progmem for some reason?

I tried other examples in the library, but I seem to be getting the same issue no matter what I try and do.

I tried backtracking libraries etc but this hasn't changed anything which has led me to believe this might be something they have updated in the IDE.

Is there a way I can please get this to work again as expected?

Bodmer, 2019-06-02:
It is likely you are using setAddrWindow in your code which has had a parameter change to make it compatible with Adafruit_GFX.

Post your drawIcon() function so I can see it or look at the new library examples that use pushImage().

clivemist, 2019-06-02:
Thanks for getting back to me.

That's great, I've now swapped out offending lines that were causing the issue, and the images are now once again straight.

Another slight issue now, it seem that red and blue information is inverted (at least that's what I am guessing as black and white are still correct).

I tried to change the line/s:

`// #define TFT_INVERSION_ON
// #define TFT_INVERSION_OFF`

Which showed me that there was no inversion but (as a guess) a colour ordering issue.

I then attempted to change:

`//  #define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
//  #define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red`

This had no apparent effect (I can see that this is probably only effective for the ST7789, although inverting the image works in the previous lines.

Very nearly there!

If you could please let me know how to resolve the colour reordering that would be amazing.

That's so much for this library! It is an absolute gem!

I am using an ESP32Wroom with an ILI9341.

Bodmer, 2019-06-02:
This may be due to the byte ordering in the image. Try adding [this](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Generic/TFT_Flash_Bitmap/TFT_Flash_Bitmap.ino#L46) line to your sketch

clivemist, 2019-06-03:
That's the one!

I just added it in as "true" and everything is as expected again now.

Thanks for all the help!

closed 2019-06-03 by clivemist