Bodmer/TFT_eSPI #373: ESP32 1.0.3-rc1 Compilation error

KuhnDaniel, 2019-06-04:
In file included from /home/daniel/sketchbook/libraries/TFT_eSPI/Fonts/glcdfont.c:6:0,
                 from /home/daniel/sketchbook/libraries/TFT_eSPI/TFT_eSPI.h:62,
                 from /home/daniel/sketchbook/libraries/TFT_eSPI/TFT_eSPI.cpp:17:
/home/daniel/sketchbook/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'virtual int16_t TFT_eSPI::drawChar(uint16_t, int32_t, int32_t, uint8_t)':
/home/daniel/sketchbook/libraries/TFT_eSPI/TFT_eSPI.cpp:4273:120: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
       flash_address = pgm_read_dword( (const void*)pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *) );
                                         ^
/home/daniel/sketchbook/libraries/TFT_eSPI/TFT_eSPI.cpp:4273:120: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
       flash_address = pgm_read_dword( (const void*)pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *) );
                                         ^
home/daniel/sketchbook/libraries/TFT_eSPI/Extensions/Sprite.cpp: In member function 'virtual int16_t TFT_eSprite::drawChar(uint16_t, int32_t, int32_t, uint8_t)':
/home/daniel/sketchbook/libraries/TFT_eSPI/Extensions/Sprite.cpp:1682:120: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
       flash_address = pgm_read_dword( (const void*)pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *) );
                                         ^

home/daniel/.arduino15/packages/esp32/hardware/esp32/1.0.3-rc1/cores/esp32
/pgmspace.h:45:10: note: in definition of macro 'pgm_read_dword'
   typeof(addr) _addr = (addr); \
              ^
/home/daniel/sketchbook/libraries/TFT_eSPI/Extensions/Sprite.cpp:1682:120: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
       flash_address = pgm_read_dword( (const void*)pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *) );
                                         ^

KuhnDaniel, 2019-06-05:
it is really funny. immediately after opening the IDE during the first compilation, the error messages will appear. Also with other libraries. If you continue to compile, you will no longer receive error messages. ?!

Bodmer, 2019-06-11:
This is a "warning" so the code still compiles. The warning does not appear again (because the compiled code is cashed) unless that code section has to be recompiled, for example changing the COM port will make this happen or changing the library code.

I will get rid of the warning in the next release.

Thanks for reporting this.

closed 2019-06-11 by Bodmer