Bodmer/TFT_eSPI #375: readRect() and pushRect() changing pixel colours

egzplicit, 2019-06-12:
First of all thank you for all the hard work you've put into this project!

I'm seeing an issue where pixels get weird when using the `readRect()` and `writeRect()` methods.  I need to display a "popup" message on a ILI9341 display (SPI). Because the display can show anything at the time the popup needs to be rendered, I use `readRect()` to store the pixels underneath the popup, then render the popup and message.

Once a certain amount of time has passed, I'm calling `writeRect()` to get the display to show the previous state, before the popup was shown.

This works great but it seems the pixels somehow are changed, please see the following 2 images, one before and one after the `writeRect()` (in the first image the dotted square is where the popup was roughly shown):

[image]

The main part of the display is a jpeg file that I render at the beginning, with the numbers updating after. This is the code that renders the jpeg, but I don't think that would cause the issue because `readRect()` should read what's on the display at a pixel level as far as I understand.

```
  if (!SPIFFS.exists(filename)) {
    return;
  }

  fs::File jpegFile = SPIFFS.open(filename, "r");
  boolean decoded = JpegDec.decodeFsFile(filename);

  if (!decoded) {
    return;
  }

  uint16_t *pImg;
  uint16_t mcu_w = JpegDec.MCUWidth;
  uint16_t mcu_h = JpegDec.MCUHeight;
  uint32_t max_x = JpegDec.width;
  uint32_t max_y = JpegDec.height;

  bool swapBytes = _tft->getSwapBytes();
  _spr->createSprite(max_x, max_y);
  _spr->setSwapBytes(true);

  // Jpeg images are draw as a set of image block (tiles) called Minimum Coding Units (MCUs)
  // Typically these MCUs are 16x16 pixel blocks
  // Determine the width and height of the right and bottom edge image blocks
  uint32_t min_w = min(mcu_w, max_x % mcu_w);
  uint32_t min_h = min(mcu_h, max_y % mcu_h);

  // save the current image block size
  uint32_t win_w = mcu_w;
  uint32_t win_h = mcu_h;

  // save the coordinate of the right and bottom edges to assist image cropping
  // to the screen size
  max_x += 0;
  max_y += 0;

  // Fetch data from the file, decode and display
  while (JpegDec.readSwappedBytes()) {    // While there is more data in the file
    pImg = JpegDec.pImage ;   // Decode a MCU (Minimum Coding Unit, typically a 8x8 or 16x16 pixel block)

    // Calculate coordinates of top left corner of current MCU
    int mcu_x = JpegDec.MCUx * mcu_w + 0;
    int mcu_y = JpegDec.MCUy * mcu_h + 0;

    // check if the image block size needs to be changed for the right edge
    if (mcu_x + mcu_w <= max_x) win_w = mcu_w;
    else win_w = min_w;

    // check if the image block size needs to be changed for the bottom edge
    if (mcu_y + mcu_h <= max_y) win_h = mcu_h;
    else win_h = min_h;

    // copy pixels into a contiguous block
    if (win_w != mcu_w) {
      uint16_t *cImg;
      int p = 0;
      cImg = pImg + win_w;
      for (int h = 1; h < win_h; h++) {
        p += mcu_w;
        for (int w = 0; w < win_w; w++) {
          *cImg = *(pImg + w + p);
          cImg++;
        }
      }
    }

    // draw image MCU block only if it will fit on the screen
    if (( mcu_x + win_w ) <= _spr->width() && ( mcu_y + win_h ) <= _spr->height())
      _spr->pushImage(mcu_x, mcu_y, win_w, win_h, pImg);
    else if ( (mcu_y + win_h) >= _spr->height())
      JpegDec.abort(); // Image has run off bottom of screen so abort decoding
  }

  // push sprite to screen
  _spr->pushSprite(xpos, ypos);
  _spr->setSwapBytes(swapBytes);
  _spr->deleteSprite();
```

(_spr is just a sprite that I use to speed things up a bit)

egzplicit, 2019-06-12:
The array used to store the result of `readRect()` is in PSRAM but I've tried using the normal RAM with the same issue still showing. This is the code that I use to read/write that area:

```
uint16_t* screenContent = NULL;
screenContent = ( uint16_t*) ps_calloc(modalWidth * modalHeight, sizeof(uint16_t));
_tft->readRect(paddingLeftRight, paddingTopBottom, modalWidth, modalHeight, screenContent);

// ... draw all the things, delay a bit and then

_tft->pushRect(paddingLeftRight, paddingTopBottom, modalWidth, modalHeight, screenContent);
free(screenContent);
```

Bodmer, 2019-06-18:
I will setup one of my displays and run some tests. I suspect a bit shift is happening during the read, have you tried reducing the read SPI clock frequency?

Bodmer, 2019-06-23:
I have run some tests and I suspect your read SPI frequency is too high. Try 10MHz for the read frequency and a maximum of 40MHx for write, my setup is:

#define SPI_FREQUENCY            40000000
#define SPI_READ_FREQUENCY  10000000

My test sketch is attached. At 20MHz the TFT cannot provide the data fast enough and this is above the data sheet specifications.
[TFT_Read_Write.zip](https://github.com/Bodmer/TFT_eSPI/files/3318636/TFT_Read_Write.zip)

retrackio, 2019-06-23:
Thank you for looking into it. I remember I tried lowering the read speed thinking that might cause the issue but I didn't see any improvement.

I'll give it another go and report back.

egzplicit, 2019-06-23:
I remember trying to lower the read speed thinking that might cause the issue but I didn't see any improvements.

I'll play with both and check the outcome and report back, thanks for looking into it.

Bodmer, 2019-06-23:
The example I attached just reads the screen and writes back the same patch. I saw no corruption after 1000's of read/writes. Maybe try the test sketch.

closed 2019-07-05 by Bodmer

egzplicit, 2019-07-12:
Hi @Bodmer , 

Sorry for the slow replies, I did manage to test it with a lower read speed (`DSPI_READ_FREQUENCY=10000000`) and it does indeed fix the issue indeed.

It's funny because I did think the speed might be an issue but I changed it in the `User_Setup.h` file forgetting I have it configured via CPPFLAGS...

Thanks again for looking into this and for the work on this library!

Bodmer, 2019-07-13:
Glad to hear you have it working. Thanks for telling me.