Bodmer/TFT_eSPI #379: Feature request - support for ST7789 1.14" 135x240 TFT

BbIKTOP, 2019-06-14:
There's a lot of 135x240 TFT screens available on ebay and aliexpress. They do work with ST7789 driver, except they have different colstart and rowstart values. It's colstart = 52 and rowstart = 40 for rotation 0 and so on accordingly. Is it possible to add support for such a displays? Currently I have modified standard headers changing ST7789_Rotation.h but it will be nice probably to move it to the dedicated driver. It's good that width and height can be changed via #define but these offsets cannot, as far as I understand the code.
Thank you.

```
$ diff ST7789_135_Rotation.h ST7789_Rotation.h
8,9c8,9
<       colstart = 52;
<       rowstart = 40;
---
>       colstart = 0;
>       rowstart = 0;
19,20c19,20
<       colstart = 40;
<       rowstart = 52;
---
>       colstart = 0;
>       rowstart = 0;
30,31c30,31
<        colstart = 52;
<        rowstart = 40;
---
>        colstart = 0;
>        rowstart = 80;
40,41c40,41
<       colstart = 40;
<       rowstart = 52;
---
>       colstart = 80;
>       rowstart = 0;
```

Bodmer, 2019-06-18:
Adding configurable offsets is on my to do list but it may be quite a while before I can find time to make and test the changes.

BbIKTOP, 2019-07-07:
Sure, but what about a driver for it? I suppose it would help people like me, who bought this screen

Bodmer, 2019-07-07:
I do not have one of those displays to test the offsets.  I suspect that the offsets in the list you gave are not correct, in two places 52 should be 53.  This is because 52+235+53 = 240. You can check by drawing a rectangles in the periphery pixels (135x240 and 240x135) for each rotation and seeing if a row is missing.

I suspect the correct offsets are:
Rotation 0
       colstart = 52;
       rowstart = 40;
Rotation 1
       colstart = 40;
       rowstart = 53;
Rotation 2
        colstart = 53;
        rowstart = 40;
Rotation 3
       colstart = 40;
      rowstart = 52;

BbIKTOP, 2019-07-08:
Yes, you're absolutely right. This one is correct, checked it with drawrectangle:

```
switch (rotation) {
    case 0: // Portrait
#ifdef CGRAM_OFFSET
      colstart = 52;
      rowstart = 40;
#endif
      writedata(TFT_MAD_COLOR_ORDER);

      _width  = _init_width;
      _height = _init_height;
      break;

    case 1: // Landscape (Portrait + 90)
#ifdef CGRAM_OFFSET
      colstart = 40;
      rowstart = 53;
#endif
      writedata(TFT_MAD_MX | TFT_MAD_MV | TFT_MAD_COLOR_ORDER);

      _width  = _init_height;
      _height = _init_width;
      break;

      case 2: // Inverter portrait
#ifdef CGRAM_OFFSET
       colstart = 53;
       rowstart = 40;
#endif
      writedata(TFT_MAD_MX | TFT_MAD_MY | TFT_MAD_COLOR_ORDER);

      _width  = _init_width;
      _height = _init_height;
       break;
    case 3: // Inverted landscape
#ifdef CGRAM_OFFSET
      colstart = 40;
      rowstart = 52;
#endif
      writedata(TFT_MAD_MV | TFT_MAD_MY | TFT_MAD_COLOR_ORDER);

      _width  = _init_height;
      _height = _init_width;
      break;
  }
```

test loop:

```
void loop()
{
  static int rotation = 0;

  tft.setRotation(rotation);
  tft.fillScreen(TFT_BLACK);
  if (rotation % 2)
  {
    tft.drawRect(0, 0, 240, 135, TFT_WHITE);
  }
  else
  {
    tft.drawRect(0, 0, 135, 240, TFT_WHITE);
  }
  tft.setCursor(5, 5);
  tft.println("Rotation " + String(rotation));
  Serial.println("Rotation " + String(rotation));
  if (++rotation > 3)
    rotation = 0;
  delay(5000);
}

```

Observed result - white border around screen on all its sides

Bodmer, 2019-07-08:
OK, this is helpful info. I can create a driver set for you but this wil be a temporary solution as eventually the library will allow offsets to be defined for each rotation in the setup. Due to work commitments it will be a few days before I have time to create the driver for you to test.

BbIKTOP, 2019-07-09:
Thank you. Just in case - I made a working driver, not sure is it correct, but it works on all examples you include as well as with some I made myself. Here's an archive that also contains a small video confirming everything works fine https://tut.etogo.net/in/tft_espi.135x240.st7789v.tgz

tablatronix, 2019-07-28:
Really gotta get setup offsets in the library heh this is going to get hairy

TotallyInformation, 2019-07-28:
Relates to [Issue #1](https://github.com/Xinyuan-LilyGO/TTGO-T-Display/issues/1) in the Xinyuan-LilyGO/TTGO-T-Display repo.

tablatronix, 2019-07-30:
fyi

https://github.com/lewisxhe/TFT_eSPI/commit/d594f6729510600e97d5b10cf9d8271d1fd3f46b

tablatronix, 2019-07-30:
Hmm I can confirm this works, but I cannot seem to get the display to rotate..

Bodmer, 2019-07-31:
I have made some changes and added a new user setup file "Setup135_ST7789.h" that shows how to define the required setup.

Let me know if this works OK for you.

closed 2019-08-10 by Bodmer