Bodmer/TFT_eSPI #380: ILI9486 8 bit & touch

kuzmanzs, 2019-06-15:
Hi,

I have ESP32 Uno Style board. I have done IO15 wired to IO35, IO33 wired to IO34, IO32 wired to IO36 wireing.
I am using #define ESP32_PARALLEL and #define ILI9488_DRIVER and dispaly looks good.

I am unable configure with touch sensing. I checked repo https://github.com/s60sc/Adafruit_TouchScreen and I made  pin 4 and analog pin 39 wireing as well. 
I am running https://github.com/s60sc/Adafruit_TouchScreen/blob/master/examples/ESP32testTouch/ESP32testTouch.ino but no touching detected. 

I have following display
https://es.aliexpress.com/item/3-2-inch-TFT-LCD-screen-module-Ultra-HD-320X480-for-Arduino-MEGA-2560-R3-Board/32605255967.html?spm=a219c.10010108.1000016.1.474b1359PHPlgq&isOrigTitle=true

pls help

Bodmer, 2019-06-18:
The library only supports SPI touch screens based on the XPT2046. Typically the 8 bit screens have a resistance sending interface that must be read using an ADC.

You could try the library [here](https://github.com/s60sc/Adafruit_TouchScreen).

kuzmanzs, 2019-06-18:
I have tryied this library already..

See above:
"I am running https://github.com/s60sc/Adafruit_TouchScreen/blob/master/examples/ESP32testTouch/ESP32testTouch.ino but no touching detected."

Could you give me hint what can be wrong?

Bodmer, 2019-06-18:
I have not tried that library do I am not in a position to help. I suggest you raise an issue on the libraries repository and hopefully the author will help.

closed 2019-07-05 by Bodmer