Bodmer/TFT_eSPI #382: TFT_eSPI not working together with IRremoteESP8266

Shockwave2k, 2019-06-20:
Hi,

recently i was trying to build a small home control unit which includes a Tempratur/Humidity sensor, a TFT Display and a IR sensor. Sadly its not working cause the IR Libary stops working as soon as i load the TFT libary.

I already had a conversation/open issue in the IRremoteESP8266 Repositiry which you can find here: https://github.com/markszabo/IRremoteESP8266/issues/762

It was mentioned that this migh be a timer/interrupt issue thats why he suggested to ask for it here as you will habe a more detailed overview about your libary.

> I suggest you take it up/report an issue with the TFT_eSPI library. They might know if it uses some timer or plays with or disables the interrupts on other pins etc.
> 
> Here are the major references in our code to the ESP8266 timer I can think of:
> 
> https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRrecv.cpp#L47
> https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRrecv.cpp#L84
> https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRrecv.cpp#L117
> https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRrecv.cpp#L213
> https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRrecv.cpp#L62
> 
> 
> I could probably come up with some HORRIBLE kludge to make our stuff work without a timer, but it would require your code to essentially pretend to operate like a timer/interrupt etc.
> e.g. every time you go through your main loop, check if it has started capturing something, then wait for N milliseconds (or do something) to see if more has been added, if not, tell it finish/decode, otherwise keep waiting until nothing gets added. It'd be completely unsupported & experimental etc, but might get you out of your bind.
> 
> _Originally posted by @crankyoldgit in https://github.com/markszabo/IRremoteESP8266/issues/762#issuecomment-503913626_

Do you have any idea what could cause this issue? Any help is appreciated and if you want me to test things or experimentel code just let me know and i will give it a try.

Bodmer, 2019-06-20:
The TFT_eSPI library does not use timers or interrupts so I am not sure where the incompatibility lies.

Have you tried using tft.begin() at the start of setup() and then initialising the IR library afterwards?

Shockwave2k, 2019-06-20:
Tried it like this now: https://gist.github.com/Shockwave2k/b4db2285a66cb4818afb55ed83ed6947 but its also not working 😞

crankyoldgit, 2019-06-21:
If this library doesn't use interrupts or timers, then that leaves the DHT lib, or the SPI lib.

I had a skim over https://github.com/beegee-tokyo/DHTesp and it doesn't appear to have anything that would affect anything `IRremoteESP8266` is doing.

Just confirming the obvious, you are doing this on a ESP8266 right? (our support of ESP32 is fairly new)

crankyoldgit, 2019-06-21:
Hmmm. DHTesp uses `cli()` and `sei()` when reading the sensor. So it does play with interrupts. Have you tried your code with the IR & TFT stuff, but without DHT?

Shockwave2k, 2019-06-21:
Yes i'm using a ESP8266.
If it would be related to the DHT this should also not work https://gist.github.com/Shockwave2k/876280ec041b09559d449874309c2041 but it does.
But i will give it another try without DHT but with TFT as soon as i'm back on my PC.

Bodmer, 2019-06-23:
In your first example you are allocating D2 to both the DHT and the IR receive library, these should be on different pins. In the second "working" example you use pin D6 for the IR receive library...

Shockwave2k, 2019-06-24:
I dont know what i have changed... but somehow it now works...
will close it for now and post an update if i find out what caused the issue.

closed 2019-06-24 by Shockwave2k