Bodmer/TFT_eSPI #384: Missing character Celsius

cawena, 2019-06-22:
The character Celsius (°) is missing. How to add this character on FONT1 and FONT2?

Thx for help

Bodmer, 2019-06-23:
The fonts only include characters up to ASCII 127 so the Celsius symbol would not normally be included in that range.

However for font 4 I made the "grave" accent into a Celcius symbol, so you can write  `C for example.

The grave accent is normally on the key next to the 1 on a keyboard.

Font 2 does not contain a Celsius symbol.

Bodmer, 2019-06-23:
If you replace Font16.h in the Fonts folder of the library with the attached then the grave accent will be a Celsius symbol in font 2 as well.

[Font16.zip](https://github.com/Bodmer/TFT_eSPI/files/3318538/Font16.zip)

closed 2019-06-23 by Bodmer