Bodmer/TFT_eSPI #385: ESP32 hangs on setFreeFont function

abasir, 2019-06-25:
Hello all,

I'm trying out PlatformIO few days ago and imported my existing Arduino-based codes.  All seem to be working fine, including all TFT_eSPI functionality.  The only hickup I'm facing now is that the setFreeFont function cause the ESP to freeze.  

Compiling the All_Free_Fonts_Demo gave the following warnings:

In file included from C:/Users/EVD-LPT-334/.platformio/lib/TFT_eSPI_ID1559/Fonts/glcdfont.c:6:0,ompiling .pioenvs\esp32doit-devkit-v1\FrameworkArduino\WString.cpp.o
from C:\Users\EVD-LPT-334\.platformio\lib\TFT_eSPI_ID1559\TFT_eSPI.h:62,
from C:\Users\EVD-LPT-334\.platformio\lib\TFT_eSPI_ID1559\TFT_eSPI.cpp:17:
C:\Users\EVD-LPT-334\.platformio\lib\TFT_eSPI_ID1559\TFT_eSPI.cpp: In member function 'virtual int16_t TFT_eSPI::drawChar(uint16_t, int32_t, int32_t, uint8_t)':
C:\Users\EVD-LPT-334\.platformio\lib\TFT_eSPI_ID1559\TFT_eSPI.cpp:4273:120: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
flash_address = pgm_read_dword( (const void*)pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *) );
C:/Users/EVD-LPT-334/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:45:10: note: in definition of macro 'pgm_read_dword'
typeof(addr) _addr = (addr); \
C:\Users\EVD-LPT-334\.platformio\lib\TFT_eSPI_ID1559\TFT_eSPI.cpp:4273:120: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
flash_address = pgm_read_dword( (const void*)pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *) );
C:/Users/EVD-LPT-334/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:45:25: note: in definition of macro 'pgm_read_dword'
typeof(addr) _addr = (addr); \
C:\Users\EVD-LPT-334\.platformio\lib\TFT_eSPI_ID1559\Extensions/Sprite.cpp: In member function 'virtual int16_t TFT_eSprite::drawChar(uint16_t, int32_t, int32_t, uint8_t)':
C:\Users\EVD-LPT-334\.platformio\lib\TFT_eSPI_ID1559\Extensions/Sprite.cpp:1682:120: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
flash_address = pgm_read_dword( (const void*)pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *) );
C:/Users/EVD-LPT-334/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:45:10: note: in definition of macro 'pgm_read_dword'
typeof(addr) _addr = (addr); \
C:\Users\EVD-LPT-334\.platformio\lib\TFT_eSPI_ID1559\Extensions/Sprite.cpp:1682:120: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
flash_address = pgm_read_dword( (const void*)pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *) );
C:/Users/EVD-LPT-334/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:45:25: note: in definition of macro 'pgm_read_dword'
typeof(addr) _addr = (addr); \

Anyone has any idea how to solve this?

abasir, 2019-06-25:
Closing this for now.  Testing with latest version of TFT-eSPI first.  Was using 1.4.8 instead of 1.4.11

abasir, 2019-06-25:
Same error with latest codes.  Take note that the same code/library worked without a hitch when compiled via Arduino IDE.  It has to be related to PlatformIO.

Bodmer, 2019-06-26:
Yes, there have been changes to the ESP32 core that affects the argument types.

Substitute the following for the lines in error:

'''
      flash_address = pgm_read_dword( &(fontdata[font].chartbl ) ) + uniCode*sizeof(void *);
      #ifdef ESP32
        flash_address = pgm_read_dword( flash_address );
      #else
        flash_address = pgm_read_dword( (const void*) flash_address );
      #endif
'''

abasir, 2019-06-27:
Thanks for the quick reply.  
Replaced the above codes in 2 locations (TFT_eSPI.cpp & sprite.cpp).
The All_FreeFonts_Demo compiled without warning/errors and worked as expected.

Nonetheless, my application still hangs at the same SetFreeFont line.  Something else must be causing it.  
Let me spend some more time with my codes and will revert if needing your help.

Thanks for the update and your time.  Closing the issue for now.  TFT_eSPI rocks!!!

closed 2019-06-27 by abasir

Bodmer, 2019-06-28:
Don't forget to call yield() or delay(0) periodically in long duration loops to stop the watchdog timing out.

Post a copy of your failing code if you need help.

abasir, 2019-06-28:
Thanks for the heads up...  yes, plenty of that in my codes...

Actually, I'm doing this project migration (Arduino IDE to PlatformIO) more to familiarise myself with PIO.  As mentioned, the project (around 3k code lines) compiled and ran without any issue when using Arduino IDE.  Compiling with PIO gave many warnings because some of the libraries I used are not yet supported by PIO.  I'm interested with PIO as PIO lately allows debugging with ESP-Prog (via JTAG) without having paid subscription.  

Now, awaiting arrival of my cheap ESP-Prog from China...