Bodmer/TFT_eSPI #386: Support for ESP32 and 9bit ST7789H2

shr1Khr, 2019-06-29:
HI Bodmer,

I was wondering if you could add or help me to add the support for ESP32 and ST7789H2 LCD with 9 bit as command/Parameter length and RGB of 27 bits total (since each is 9 bit).
I have written the SPI code for writing to the LCD from ESP32, and the read register of the LCD is via the bidirectional SDA/MOSI as implemented in your library, and reuires bit bangging.
The LCD uses IM: 0101 interface, in ST7789V/ST7789H2 datasheet.
Here is the code for your reference - 

`
uint8_t flipByte(uint8_t val){
  uint8_t r=0;
  for(int i = 0; i < 8; i++){
    r <<= 1;
    r |= val & 1;
    val >>= 1;
  }
  return r;
  }

void command(uint8_t val){  
  uint8_t v = flipByte(val);
  uint16_t vv;
  uint16_t vv1 = v;
  uint16_t vv2 = 0000;  
  vv = vv1<<1 | vv2 ;  
  hspi.transferBits(vv, NULL, 9);
}

void param(uint8_t val){
  uint8_t v = flipByte(val);
  uint16_t vv;
  uint16_t vv1 = v;
  uint16_t vv2 = 1;  
  vv = vv1<<1 | vv2 ;  
  hspi.transferBits(vv, NULL, 9);
}

// since transferBits is very expensive, so I clubbed the RGB write together
void param32(uint8_t RR, uint8_t GG,uint8_t BB){  
  uint8_t RRR = flipByte(RR);
  uint8_t GGG = flipByte(GG);
  uint8_t BBB = flipByte(BB);

  uint32_t vv1R = RRR;
  uint32_t vv1G = GGG;
  uint32_t vv1B = BBB;
 
  uint32_t vv2 = 1;  
  uint32_t vvR = vv1R<<1 | vv2;
  uint32_t vvG = vv1G<<1 | vv2;
  uint32_t vvB = vv1B<<1 | vv2;
  uint32_t vv = vvR | vvG<<9 | vvB <<18 ;  
  hspi.transferBits(vv, NULL, 27);  
  }
`
Two questions I have regarding your library and the reason I couldn't add my code is - 
1. The length of the color - i.e. in this case, the color length should be defined is 32 bit, even though we send only 27 bits. Not sure, but like in littlevgl library, shouldn't we also have the choice to select the color length in the setup section of the library?
2. How to implement the 9 bit command and parameter write in the library

shr1Khr, 2019-06-29:
forgot to add these details for spi-
hspi.setHwCs(false);
hspi.setBitOrder(LSBFIRST);

Bodmer, 2019-07-05:
The changes needed would affect many parts of the library including all the 16 bit colour handling and the SPI transactions handling (to retain the ESP8266 and ESP32), this would be many hours of work.  It would be possible, but there are so many cheap displays on the market that TFT_eSPI support that I am not sure the effort is worth it unless there is a significant benefit with the display you have.

The TFT_eSPI library master does not support 9 bit transactions but I think some users have created forks and added 7 bit support. The disadvantage of using a forked variant is that some were forked long ago and later/future updates to the master library can be tricky to add in later.

closed 2019-07-05 by Bodmer