Bodmer/TFT_eSPI #387: ESP32 with ILI9341 8-bit display in SPI

danliberato, 2019-06-29:
So, I bought [this ILI9341display](http://www.lcdwiki.com/3.2inch_16BIT_Module_ILI9341_SKU:MRB3205#Product_Documentation) and managed to work in 8-bit parallel very well (plus SD and touch, so fisrt of all, thank you).

But as ESP32 has a low pin count, I was wondering if I can add a 74HC595 shift register (already in hands) and make things work in SPI mode.
As you can see there are CS/RS/WR/RD pins that I don't know where to wire while using SPI (both ESP32 and 73HC595 sides).

So here come the questions: Am I overthinking? Is it really possible to get done? Is an library enhancement?

Ps: already checked #299, #294, #362, #345 (especially this one), #336, and [this link](http://jumptuck.com/2011/11/03/how-to-drive-595-shift-registers-with-avr-hardware-spi/) which explains how to control a SR via SPI. But always controlling a simple led array.

Thank you!

mlongo95, 2019-07-01:
Looks like you purchased a 8-bit/16-bit Parallel Panel, you need to purchase a panel that has an SPI interface, it will make things a lot easier. Heres the same panel (ILI9341) but with SPI output [https://www.amazon.com/Display-Module-240320-4-Wire-Without/dp/B07KPDVY6T/ref=sr_1_9?keywords=3.2+tft+lcd&qid=1562013763&s=gateway&sr=8-9](url) 

Hope this helps

danliberato, 2019-07-01:
Yeah, I understand the differences between each protocol, as the fact that I got a (almost) non compatible version of this LCD with my ESP32.
My main question is: if the library can be enhanced to support SPI in a "8 bit + shift register" situation.
As I discussed [here](https://forum.arduino.cc/index.php?topic=623668.0), buying a new SPI LCD is not an option because I want to keep a low-budget project (and it would take a long time to ship).

In this moment (while the author does not give me a negative answer) I'm trying to understand the library details to see if it's possible.

Anyway, I appreciate the answer (even if it tells me that I'm on the wrong track).

Bodmer, 2019-07-05:
The library could be modified but this is not something I have time to do for a unique project.

A few pins can be freed, for example CS stays low all the time and thus that pin can be re-purposed, also the read line stays high unles the display is read by the user sketch so again that pin can be re-purposed and used by the sketch, clearly these pins would need to be disconnected from the TFT and the TFT pins pulled to the required state.

The data pins could be re-purposed via an additional latch by using the freed control lines as a user controlled chip select and strobe. It all depends what you want to do with the extra pins...

closed 2019-07-05 by Bodmer

danliberato, 2019-07-17:
I understand that is a very specific case.
Anyway, thanks for the answer.