Bodmer/TFT_eSPI #388: Smooth Font for emojis bigger than 0xFFFF

Petros144, 2019-07-04:
Hey Bodmer & comunity!

I really want to make a emoji font for the TFT Lib, I already have the Processing IDE up & runnig with the create font prgramm.

now i need to adress a lager unicode than 0xFFFF for example 0x1F600 for the standart smiley.

but if i put this into processing it gives me only rectangles. is there a way to make this work?

Petros144, 2019-07-04:
Edit:
With this line I can Print the Emoji to the preview window, but its not in the font i want to make.

text(new String(Character.toChars(0x1F600)), width/2, height/2);

Bodmer, 2019-07-05:
TFT_eSPI supports code points up to 0xFFFF (BMP range) in monochrome so even if Processing does support higher code point values it will not be possible to use the vlw output file and access the emoji.

One option, if the emoji are monochrome, is to map the characters into the 0-FFFF range, you may be able to find a utility to do this.  There are also other true type fonts that include emoji in the required code point range.

A second option for multi-colour emoji is to turn the characters into images.

closed 2019-07-05 by Bodmer