Bodmer/TFT_eSPI #390: using different screen size (eyes example)

bradanlane, 2019-07-07:
I looked at issue #309 and I am seeing similar split image / wrapped image.

I am using a ST7789 240x240.

When I change the rotation, the issue just moves to a different quadrant of the screen.

I tried changing the `setAddrWindow()` and any values other than the example of `setAddrWindow(0,0,127,127)` results is garbage on the screen.

I hate to say *HELP* but that is the point I've reached.

My first goal is to just center the eye on the screen. Subsequently I'd like to be able to scale the eye to fill the screen.

Bodmer, 2019-07-08:
Adafruit changed the parameters to the setAddrWindow() function and though the library has been update to be Adafruit_GFX compatible some of the examples maybe have not...
The new parameters are x,y,w,h so this may work:

setAddrWindow(0,0,128,128);

To mode the image only the x and y need to be changed, e.g.

setAddrWindow(20,10,128,128);

bradanlane, 2019-07-08:
Splendid! 

I set `setAddrWindow(55,55,128,128)` and the eye is centered.

While I hate to push my luck, any chance you could give a hint as to how I might scale the eye up to use more of the 240x240 screen? (If not, then my original issue has been addressed an it can be closed).

Bodmer, 2019-07-08:
Scaling would be tricky and would impact the frame rate significantly. The easiest option for a larger eye is to use a low cost 128x128 1.4" TFT which the sketch was designed to drive  full screen, these displays have have bigger pixels.

closed 2019-07-08 by Bodmer

bradanlane, 2019-07-08:
Thank you. I have a 128x160 which I can test.

JackIsGoofingOff, 2024-10-24:
@Bodmer Is it possible to execute a global screen scale rendering with tft_espi? Really looking forward to this feature.